motioneye-project/motioneye #1873: issue with docker compose and missing config data

holzfelix, 2020-08-01:
Hi togehther, I use the following docker-compose file. 

`---
version: "3.5"
services:
  motioneye:
    image: ccrisan/motioneye:master-amd64  # Change to ccrisan/motioneye:master-armhf for ARM chips (Pi etc.)
    hostname: "motioneye"
    restart: "always"
    ports:
      - "8081:8081"
      - "8765:8765"
    volumes:
      - /etc/motioneye:/etc/motioneye
      - /var/lib/motioneye:/var/lib/motioneye
      - /media/CamData:/media/CamData`

It worked fine an hour ago^^ ... then I restartet the PC because I moved it to the cellar ... and now I got this error:

>     INFO: hello! this is motionEye server 0.42
> 
> 
> CRITICAL: config directory "/data/etc" does not exist or is not writable

What could be the issue in this case?

closed 2020-08-01 by holzfelix

marcusobrien, 2023-02-07:
@holzfelix - Hi - I am also getting this - did you fix this ?

holzfelix, 2023-02-07:
Nope :( dosn't work. Every Reboot all data is lost and I have to build a new docker. and import the backuped settings