motioneye-project/motioneye #1874: Motion log level not as configured in motion.conf

mcgyver83, 2020-08-02:
Hi, I'm running motioneye with docker on a rpi with image ccrisan/motioneye:dev-armhf 2b2410e11438.
Despite I configured e lower log level motion inside the container starts with WARN.

```
sudo docker exec -it motioneye /bin/sh
cat /etc/motioneye/motion.conf
# @enabled on
# @show_advanced on
# @normal_password adsasdasdas
# @admin_username admin
# @admin_password 61d1d010db589bd605332489871o691cd536987
# @normal_username user

webcontrol_port 7999
setup_mode off
webcontrol_parms 2
webcontrol_localhost on

camera camera-1.conf
camera camera-2.conf

log_level 4
log_type ALL
daemon on
native_language on

webcontrol_interface 1
```

```
sudo docker exec -it motioneye tail -f /var/log/motion.log
[0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
[0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-2.conf
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2 Started
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
```

I tried removing both image and container and starting from scratch but no luck

zagrim, 2020-08-03:
I don't know if the issue is due to that MotionEye sets Motion log level based on the log level of ME and is thus overwriting what you configure in `motion.conf`, so if you change the `log_level` in `/etc/motioneye/motioneye.conf` to `error` also Motion log level should get configured likewise.

mcgyver83, 2020-08-03:
In motioneye.conf i have INFO as log level
```
# the log level (use quiet, error, warning, info or debug)
log_level info
```

So if motioneye overwrites motion.conf setting it should starts with INFO 😞

zagrim, 2020-08-03:
Do the individual camera config files (`camera-1.conf` etc) have `log_level` mentioned? That would override whatever set globally, I think. Other than that, I can't think of any explanation for this.

mcgyver83, 2020-08-03:
no log directive except in motion.conf and motioneye.conf

```
motioneye/conf $ grep -r "log" .
Binary file ./core matches
./motioneye.conf:# path to the directory where log files go (must be writable by motionEye)
./motioneye.conf:**log**_path /var/**log**
./motioneye.conf:# the **log** level (use quiet, error, warning, info or debug)
./motioneye.conf:**log**_level info
./motion.conf:**log**_level 4
./motion.conf:**log**_type ALL
```

zagrim, 2020-08-03:
I don't seem to find a way to match the docker image digest to a ME version, so: What's the MotionEye version you are running? I seem to remember there once was a off-by-one error in the mapping of log level from ME to Motion, but didn't find the corresponging issue or fix, though.

Have you tried setting `log_level error` in `motioneye.conf` to see if that makes any difference?

zagrim, 2020-08-03:
Anyway, I just tested on the local development setup I have (with a bleeding edge version) and the log level of Motion beatifully follows the log level of Motioneye. I don't have `log_level` in `motion.conf`.

mcgyver83, 2020-08-03:
Please could you share motion.conf and motioneye.conf of a fresh install?
If I red correctly motion.conf is copied from a local default copy inside the container into the config folder (docker volume mounted)

zagrim, 2020-08-04:
I attached initial `motioneye.conf` and `motion.conf` after adding a camera. Before adding a camera there was no `motion.conf` at all.
[initial_config.zip](https://github.com/ccrisan/motioneye/files/5019816/initial_config.zip)

mcgyver83, 2020-08-04:
I have motion.conf and motioneye.conf like yours.
I'm running motioneye inside docker container.

I still have these lines in motion.log inside the container
```
[0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
```

Please could you check your motion.log file to verify that your motion is starting using log level INFO (in your motioneye.conf log_level si INFO)

zagrim, 2020-08-04:
Yes, in my case the log level of Motion is indeed INFO (I can see INFO level messages being logged).
I don't know if there are some peculiarities with the docker install, I haven't ever done that myself.

mcgyver83, 2020-08-05:
This is my docker run script:
```
#!/usr/bin/env bash

docker stop motioneye; 
docker rm motioneye; 
docker volume prune -f;
docker pull ccrisan/motioneye:dev-armhf;
docker run --name="motioneye" \
	    -p 9000:8765 \
		-p 9001:9001 \
		-p 9002:9002 \
		-p 9003:9003 \
		-p 9004:9004 \
		-p 9009:7999 \
		--env="PUID=1234" \
    	        --env="PGID=1234" \
		--hostname="motioneye" \
		-v /etc/localtime:/etc/localtime:ro \
		-v /home/xpz/local_drive/docker_motioneye/conf:/etc/motioneye:rw \
		-v /home/xpz/local_drive/motioneye_storage:/var/lib/motioneye:rw \
		--restart="unless-stopped" \
		--detach=true \
		ccrisan/motioneye:dev-armhf;
```

Could you try with this ?

zagrim, 2020-08-06:
I looked already earlier into the log level mapping code in meyectl.py but it seems I failed to count the Motion log levels correctly, as there's sort of a mismapping there when selecting the value for `-d` command line parameter of Motion.
```
   if settings.LOG_LEVEL <= logging.DEBUG:
        args.append('9')

    elif settings.LOG_LEVEL <= logging.WARN:
        args.append('5')

    elif settings.LOG_LEVEL <= logging.ERROR:
        args.append('4')

    else:  # fatal, quiet
        args.append('1')
```
So, ME log level debug maps to "all" ("more than debug"),  warning and info both map to warning, error maps to error, and quiet maps to emergency.

I confirmed this by looking at my setups and saw that the log level of Motion is set via command line parameter (in my case `/usr/local/bin/motion -n -c /etc/motioneye/motion.conf -d 5` with `log_level info` in motioneye.conf). This also means you can't override the log level of Motion by setting anything in the conf file, as the command line argument has higher priority.
What I don't know why the log level mapping has been done like that, is that based on the behaviours so that the amount or "expressiviness" of the logging in the two applications somehow matches when mapped like that. The corresponding commit message is "motion log level follows now as much as possible motionEye's log level". Applications rarely log at all the possible levels and Motion having a 9 level scale likely exhibits that, too.

Now the question to you @mcgyver83 is whether you need to have Motion logging at "info" level all the time, or if you could just go with setting MotionEye logging to "debug" for the time you need more verbosity from Motion?

zagrim, 2020-08-06:
It would be no problem for me to create a PR for setting the log level mapping straight, but I'd prefer having @ccrisan tell his point in having made it like it now (see above comment), to avoid creating a change that is not approved in the end.

mcgyver83, 2020-08-06:
@zagrim I have some cheap ip cam (Sricam brand) and I have some issue (due crappy cam) so I'm troubleshooting motioneye and motion to understand if I can use these softwares to continue record  these camera.

zagrim, 2020-08-06:
Please try to set `log_level debug` in motioneye.conf, does that create too much logging?

mcgyver83, 2020-08-07:
Ok, now I have more logs. I'm fine now.
I saw motion.conf and motioneye.conf both with log level configuration and I thought to be able to configure them.
I thought that could be nice to ask both motion and motioneye to log all.
Right now I'm fine for my troubleshooting purpose. 

thanks

mcgyver83, 2020-08-08:
Coming back to this, in fact with debug in motioneye.conf I have a lot of information.
But could be nice to configure motion and motioneye log separately, to dig deep in motion or motioneye message.

closed 2021-03-05 by starbasessd

nooxnet, 2021-11-20:
But this has not been fixed yet. Should be an easy fix.

starbasessd, 2021-11-20:
@nooxnet There wasn't any traffic/comments since 8 August 2020 7 months at the time. I am slowly trying to clean up / close issues that have no interest or input or follow up or are outdated.