motioneye-project/motioneye #1876: IP camera address not editable in GUI?

Yonny24, 2020-08-03:
Is there a config file I can access to modify the IP address of the cameras, in the case of a camera's IP not being static? 
Currently, I need to delete the entire camera and start all over again if the IP changes as this field is not editable in the motioneye gui. Happens if I get a power cut for example.

Maybe there isa a config file I can modify but haven't found it.
I'm unable to set a static lease on my router for 3 of the cameras.

Thanks

zagrim, 2020-08-03:
Yep. In the same directory where the rest of the config is (likely `/etc/motioneye` at least for non-docker installation) you'll find files named `camera-N.conf` ("N" being the number corresponding to a camera).

I'd suggest setting the cameras to use a static IP if at all possible (if they get their IP from your router, there's likely an option to reserve IPs in the DHCP settings, or possibly the cameras have some web admin GUI where their network settings could be set.
If that isn't possible, one option to having to edit 3 different files in case all 3 change their IP is to add entries in `/etc/hosts` (on the MotionEye host) for the cameras so that you have local aliases for the cameras which you then can use in MotionEye config to refer to the cameras. The format to be used in `/etc/hosts` is simply "<IP> <alias>" (e.g. `1.2.3.4 mycam-1`), one entry per line.

Yonny24, 2020-08-03:
Hi zagrim.

I had a look in all the folders named /etc/ last night but could not find anything called motion or motioneye. (ubuntu 20).

Possibly because it is a supervisor addon?

My dhcp server is locked down by the iSP provider. Nothing I can edit there, unfortunately.

zagrim, 2020-08-04:
Are you running MotionEye in Docker, then?

zagrim, 2020-08-04:
From your other question it seems to me that you are actually running MotionEye as an addon to Hass.io, am I right? I know nothing of how things are set up in that case, or where the configs are put, but surely the installation instructions mention that.

Yonny24, 2020-08-04:
Hi zagrim. Yes that is right.
ok no worries I'll continue asking about.  Thanks for your replies.
If I can't get to the config files with the addon I may consider wiping it and installing it centrally from ubuntu instead of an addon.

zagrim, 2020-08-04:
If you can access the command line on the host running Hassio, there are some options:

- Brute force search option to find the files is running `sudo find / -name motioneye` on the Hassio host.

- Then there's the option of editing the config files inside the container by opening a shell to the MotionEye container. That could be done by first finding the MotionEye container ID with `docker ps` (the MotionEye container likely can be identified in the listing) and then issuing `docker exec -it <motioneyecontainer> bash` where `<motioneyecontainer>` is the container ID. I'm not sure if those commands require you to run them with `sudo` or not. Unfortunately there's also the possibility that there's no text editor (likely options would be `nano` and `vi` or `vim`) available in the container.

- With some googling it also seems that one can check the volume mounts of a container (see e.g. https://stackoverflow.com/q/30133664/2745865) to identify what has been mapped as `/etc/` or `/etc/motioneye` in the container. This way it might probably be easier to edit the files since you could access them outside of Docker.

Good luck with the search!

Yonny24, 2020-08-04:
Thanks zagrim. When I get some more time I'll look into the options you gave me.
I also found if I download the backup file from motioneye and open it in notepad I have all the config there including the camera url/ip.
I wonder if I can update that, save it and restore back to motioneye.

zagrim, 2020-08-05:
I'd guess modifying the backup and then restoring from it would work, too, although I've not familiar with the content/format of the backup.

closed 2021-03-05 by starbasessd

LanceMcCarthy, 2022-06-11:
In case it helps for future visitors, I thought I'd add a comment.

I am running motioneye in a docker container (specifically, in Home Assistant). I found the camera configs in `/data/motioneye` folder.

CasperJohansen, 2023-08-10:
> I'd guess modifying the backup and then restoring from it would work, too, although I've not familiar with the content/format of the backup.

Can confirm it works!
- Download backup from motioneye
- unzip it to folder
- change what you need to change in the .config files
- use 7zip to wrap it in TAR
- use online service like Zamzar to convert it to tar.gz
- restore from the new tar.gz
- restart motioneye in addons

CasperJohansen, 2023-08-10:
You can actually gz compress the tar in the same step:
> 
> `tar -cvzf filename.tar.gz /path/to/directory_or_file`
> 
> So let's say you downloaded a folder named "mymotioneyebackup" to your downloads folder. Open Downloads folder in command line (bash or Powershell) and use the one-liner
> 
> `tar -cvzf mymotioneyebackup.tar.gz ./mymotioneyebackup`

Nice! thanks for the tip!