motioneye-project/motioneye #1879: What can i do to improve my setup?

gardiol, 2020-08-07:
Hi all!
I have been working on my setup for many hours now... i try to describe it to you, then i will ask a couple of questions.

CPU:  Intel(R) Xeon(R) CPU E3-1535M v5 @ 2.90GHz
RAM: 16GB

Cameras:
5 AXIS 211M IP cams
4 Chinese V380 IP cams
2 DAHUA Bullet IP cams
1 Misecu full-hd IP cam

The AXIS cams are configured as http://<ip>/snapshot.jpg, at 1280x1024
The V380 cams are configured as RTSP 640x480 (using secondary stream to match resolution)
The DAHUA cams are configured as RTSP at 1280x960 
The  MISECU cam is configured as RTSP at 1920x1080 (again, to match resolution)

My system is running over 200% of CPU, and quite often the V380 cams simply do not show any image ( i get a lot of "connection refused" in the logs). I pinned down the issues to excessive CPU usage (whoa, not that hard eh!),

I did some extensive debug and checks:
- Each AXIS camera (configured as JPG snapshot) uses about 5% of CPU
- Each DAHUA/MISECU RTSP camera eats up to 15% or CPU.

Lowering resolution do help a bit, but i cannot lower the MISECU (1920x1080 fixed).

The V380 are a mistery as no settings can be changed and connecting to them is hit and miss, and seems to suck up to 30% CPU.

I do use movie passtrough...

What can i do?

Park0, 2020-08-20:
I am just playing with the same thing. I think a raspberry pi could be used for transcoding some streams (writing to a NFS share). But i have to try how many it can handle.

gardiol, 2020-11-23:
I doubt a pi can handle more than a couple of streams.
I improved a bit by enabling passtrough recording on most of the streams. 
Still not optimal, but i understand it would be a problem to fix in Motion and not in MotionEye.

closed 2020-11-23 by gardiol