motioneye-project/motioneye #1885: How i can fix "Unable to open video device"

UrbanSide, 2020-08-15:
[image]
I have this error,on Orange Pi lite, with 

```
motionEye Version | 0.42.1
Motion Version | 4.0
OS Version | Ubuntu 18.04
```
My cam translate on rtsp://admin:123456@192.168.1.10:554/mode=real&idc=1&ids=1 and i use it on cam settings in MotionEye

starbasessd, 2020-11-04:
I would update/upgrade motion & motionEye, if you haven't already.
Closing, no comments/requests/followup after 60 days
If the issue is still not resolved, feel free to re-open or open a new case.

closed 2020-11-04 by starbasessd

Kras-s-savchiK, 2021-02-13:
Can you fix this issue?

starbasessd, 2021-02-13:
Please open a new issue.
Include the following information:
Which SBC?
What base OS?
Which version of motion?
Which version of motionEye?
Which install instructions were followed?
What camera used?
How are your SBC and camera connected?