motioneye-project/motioneye #1891: Camera not write data to folder or GoogleDisk

UrbanSide, 2020-08-19:
[image]
My config motion detection.
[image]
My config file storage
When driving, it does not write files either to memory or to the cloud. What am I doing wrong or how to set it up?

Yonny24, 2020-08-19:
Try this format. I added 2 folders, MotionEye and then another sub folder Drive,
Make sure you authorize it by obtaining the key.. test and apply.

`Location: /MotionEye/Drive/`

zagrim, 2020-08-20:
Have you verified that motion is indeed detected long enough to trigger saving of video/pics? Not knowing your setup, but the Frame Change Threshold seems pretty high, and if the framerate is low, having Minimum Motion Frames at 35 might mean it needs several seconds of motion on at least 1/5th of the video frame area before saving anything.

Anyway, the easiest way to test the storage settings is by adding `emulate_motion on` in Extra Motion Options under Video Device, with that Motion will keep on pretending it detects motion.

UrbanSide, 2020-08-20:
> Have you verified that motion is indeed detected long enough to trigger saving of video/pics? Not knowing your setup, but the Frame Change Threshold seems pretty high, and if the framerate is low, [...]

I set the settings below, turned on emulation does not work. If you start the camera only through **motion**, everything works, shows a moving object.

UrbanSide, 2020-08-20:
> y this format. I added 2 folders, MotionEye and then another sub folder Drive,
> Make sure you authorize it by obtaining the key.. test and apply.
> 
> `Location: /MotionEye/Drive/`

Thanks for the answer, I did the same. Doesn't work for some reason. Apparently the point is that he does not see the movement.

zagrim, 2020-08-21:
I don't quite understand this. Do you mean that when running Motion by itself you get motion detected and files being saved? Did you use the same config files (`motion.conf` + `camera-X.conf`) MotionEye creates, or did you make any changes?
If the same config files were being used, what is the actual result with MotionEye? Any errors in the logs or in the GUI?

UrbanSide, 2020-08-21:
> I don't quite understand this. Do you mean that when running Motion by itself you get motion detected and files being saved? Did you use the same config files (`motion.conf` + `camera-X.conf`) [...]

If i start **motion -n** detection work and save date. If i start motion detect from MotionEye,its not work.Sorry for my lang i am russian.

zagrim, 2020-08-24:
I have no idea how this could make any difference when running Motion directly or via MotionEye, but in the above config there seem to be output disabled and a conflict in the threshold settings.

- There's `ffmpeg_output_movies off` which explicitly says do not save any video, likewise there's `output_pictures off` so it will not save any pics, either. Please switch the config section "Movies" to ON.

- The conflict:
```
threshold_maximum 2
threshold 42393
```
Unless I understand Motion config docs wrong, this would only trigger detection when 1 or 2 pxels change. MotionEye should probably validate the Max Change Threshold agains Frame Change Threshold to prevent such a conflict, but anyway, you probably should go with `threshold_maximum 0` first.

UrbanSide, 2020-08-28:
> I have no idea how this could make any difference when running Motion directly or via MotionEye, but in the above config there seem to be output disabled and a conflict in the threshold settings.
>
> * [...]

Not work, had to give up MotionEye in favor motion.

starbasessd, 2020-11-04:
Closing, no comments/requests/followup after 60 days
If the issue is still not resolved, feel free to re-open or open a new case.

closed 2020-11-04 by starbasessd