motioneye-project/motioneye #1892: Re-emergence of issue #134 (endless reboot loop)

ShadeTechnik, 2020-08-20:
I recently installed Motioneye in a Portainer container as described here [https://dbtechreviews.com/2020/05/how-to-install-motioneye-on-omv-and-docker/](url). This is a new method for me, I was previously running it on an ARM SBC for over a year but after upgrading my main NAS it seemed to be a better way to do it.

The portainer install worked fine for a week or 2, after a reboot one day it never started properly again. The logs show:
`INFO: hello! this is motionEye server 0.42
CRITICAL: media directory "/data/output" does not exist or is not writable`

This issue was mentioned in #158 and #134 I'm not really sure how to deal with this in a container other than to dump it and reinstall which I'd rather not have to reconfigure everything to have it come back in a week again

starbasessd, 2020-11-04:
Did you check permissions of /data/output? Does it even exist? Did you ty to change the path where it's stored (movies and pics?) Do an update?

ShadeTechnik, 2020-11-05:
I posted this quite some time ago so I do not remember the details. The directory did exist as it was in use just prior. Permissions was possibly an issue but I don't recall. I wound up re-installing the container to fix it.

closed 2020-11-05 by starbasessd

magicbycalvin, 2021-11-17:
I have been running in to the same problem. From what I've read on other threads, I believe that one of the delete functions ends up wiping out the entire /data directory. My experience is identical to the original post above and I do recall using the delete function to clear out my NAS.

I would be happy to attempt to fix this, but I have no idea where to start looking. Otherwise, I will patiently wait until this is fixed and in the meantime, I'll spin up a new container.

Thank you for your time!