motioneye-project/motioneye #1894: Feature request - Use mask filter to filter out parts of the capturing area

DkYSwe, 2020-08-20:
It would be really great if parts of the area the camera is capturing could be filtered out using the same type of filter function that is used for motion detection areas.

Initially - the filtered out area could be all grey or black but in long term it would be cool if could be blurred as option 

Keep up the good work! :+1:

zagrim, 2020-08-20:
If you mean privacy masking, it is already implemented (#1247) but seems like it didn't make it to 0.42.1 release.

DkYSwe, 2020-08-20:
> If you mean privacy masking, it is already implemented (#1247) but seems like it didn't make it to 0.42.1 release.

Great, sorry I created a dup, looking forward to try it when released :)

starbasessd, 2020-11-04:
It is in the 20200606 and dev20201026 versions of motionEyeOS.

caldir65, 2021-07-03:
Hi,
is this function only in motioneyeos, or can I use it also in motioneye installed ontop of raspios (buster) on a raspi4?

starbasessd, 2021-07-03:
Currently available as part of the motionEyeOS 20200606 and dev20201026 releases. I believe, when the next release of motionEye is completed, it should be there, too.

starbasessd, 2021-07-03:
BTW, motionEyeOS dev20201026 works on RPi4, and as an added benefit, can be booted from a USB install.

JamBalaya56562, 2026-11-08:
Closing, as this is the privacy mask from #1247, as zagrim said, which came with motionEye 0.43.1, the first release after 0.42.1. With Motion 4.1 or newer, it is under Video Device > Privacy Mask: the masked parts are blanked out in the pictures, movies and stream. Blurring instead of blanking would be a feature of Motion, which draws the mask.

closed 2026-11-08 by JamBalaya56562