motioneye-project/motioneye #1897: armhf images are listed as linux/amd64 on dockerhub

gitdeath, 2020-08-24:
This results in an error that "no node supports the platform" on arm platforms like a pi.

[image]

starbasessd, 2020-11-04:
If you are using the Install in Docker instructions you need to change the last part from 

ccrisan/motioneye:master-amd64

to

ccrisan/motioneye:master-armhf

I will update the instructions soon.

Tomcat-Engineering, 2021-02-04:
As per #1994 this happens even when you use the `-armhf` images from docker hub.  They are tagged as `linux/amd64` (see picture at the top of this ticket).  The behaviour of docker is a bit inconsistent between commands, but when it decides to check this architecture tag it complains and sometimes refuses to run the image.

I'm running docker 3.0.5 on a raspberry-pi-based industrial device and I see:

`docker pull ccrisan/motioneye:master-armhf` works.

`docker run ccrisan/motioneye:master-armhf` works.

But, for example, `docker stack deploy --compose-file docker-compose.yml` (i.e. deploying to a docker swarm) does not start any containers, and when you look at the relevant docker service you see the error `no suitable node (unsupported platform on 1 node)` like @gitdeath reported.

For anyone else experiencing this, I found that a workaround was to manually run the docker pull command above, and then add the `--resolve-image never` option to the `docker stack deploy` command.

starbasessd, 2021-02-04:
Looks like the docker-compose.yml file is at fault, it defaults to
-----
version: "3.5"
services:
motioneye:

image: ccrisan/motioneye:master-amd64 # Change to
ccrisan/motioneye:master-armhf for ARM chips (Pi etc.)
-----
and is even noted in there to change it to the armhf for Pis.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Tomcat-Engineering, 2021-02-04:
Sorry, I should have said that I'm using a custom dockerfile (that points to the armhf) image, not the one in the motioneye repo.

starbasessd, 2021-02-04:
Can you post your docker-compose.yml contents?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-02-04:
And the path where you store it? If you are not over-writing the original,
you need to include the full path to yours.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Tomcat-Engineering, 2021-02-04:
Thanks for the quick response @starbasessd.  I'm not asking for help with my docker-compose file - I was just saying that I had also seen the issue and posting my workaround in case it helped someone else.

The cause of the issue is that the arm builds on docker hub are incorrectly marked.  If you look at for example [the djaydev images on docker hub](https://hub.docker.com/r/djaydev/motioneye/tags) you'll see that they are marked for various "OS/arch" builds, whereas the [ones from this repo](https://hub.docker.com/r/ccrisan/motioneye/tags), even the ones called `armhf`, are all marked as built for `linux/amd64` as shown in the picture above, which is why people are reporting these errors and warnings from docker.

starbasessd, 2021-02-04:
What source for the Docker app install are you using?
I use
apt install docker.io
on Pis and Ubuntu/Debian VMs and don't seem to have the issue. Do you use
another method to install?
I was pointing out a possible source of the issue, but am otherwise unable
to reproduce it.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

SerialVelocity, 2021-02-04:
@starbasessd What @Tomcat-Engineering is saying is if you look here:
https://hub.docker.com/r/ccrisan/motioneye/tags?page=1&ordering=last_updated
The OS/ARCH for the `armhf` image is tagged wrongly as `linux/amd64` instead of `linux/arm/v7`. This isn't a client issue but a repo/upload issue.

(It would also be good to upload a multiarch manifest and add support for `arm64`)

Only certain docker commands trigger an error though, most just silently ignore the issue and run it anyway.

starbasessd, 2021-02-04:
@ccrisan flagging to you.to fix or close.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Bluscream, 2023-09-15:
How do i get motioneye arm64 in docker?

MichaIng, 2023-09-15:
Get the new `dev` branch Docker images from here: https://github.com/motioneye-project/motioneye/pkgs/container/motioneye

I'll mark this issue as closed, as it refers to the old DockerHub images which will be replaced with these currently hosted on GitHub registry at some point.

Bluscream, 2023-09-15:
With that version i get a lot of errors:

```
blu@minopia:~ $ sudo docker pull ghcr.io/motioneye-project/motioneye:edge
edge: Pulling from motioneye-project/motioneye
f96ab1515704: Pull complete
111a6a819be5: Pull complete
b459d6d828ec: Pull complete
562692e47d2b: Pull complete
Digest: sha256:345f7ee9501c86806ea662ff50f6185021ac22dd35b78c50f60a15ac55dbe37c
Status: Downloaded newer image for ghcr.io/motioneye-project/motioneye:edge
ghcr.io/motioneye-project/motioneye:edge
blu@minopia:~ $ sudo docker stop motioneye
motioneye
blu@minopia:~ $ sudo docker remove motioneye
motioneye
blu@minopia:~ $ docker run --name="motioneye" \
    -p 8765:8765 \
    --hostname="motioneye" \
    -v /etc/localtime:/etc/localtime:ro \
    -v /etc/motioneye:/etc/motioneye \
    -v /var/lib/motioneye:/var/lib/motioneye \
    --restart="always" \
    --detach=true \
    ghcr.io/motioneye-project/motioneye:edge
d8b0c2f20b2785da0b3f0c4dc5279cc3b149a16c350f423eeecde7f52585bc2d
```

```py
ModuleNotFoundError: No module named 'pytz'
configure_logging cmd motioneye: False
configure logging to file: None
    INFO: hello! this is motionEye server 0.43.0
Traceback (most recent call last):
  File "/usr/local/bin/meyectl", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/motioneye/meyectl.py", line 337, in main
    server.main(arg_parser, sys.argv[2:], command[:-6])
  File "/usr/local/lib/python3.9/dist-packages/motioneye/server.py", line 504, in main

)
  File "/usr/local/lib/python3.9/dist-packages/motioneye/server.py", line 425, in run
    make_media_folders()
  File "/usr/local/lib/python3.9/dist-packages/motioneye/server.py", line 344, in make_media_folders
    camera_ids = config.get_camera_ids()
  File "/usr/local/lib/python3.9/dist-packages/motioneye/config.py", line 426, in get_camera_ids
    if get_camera(camera_id):
  File "/usr/local/lib/python3.9/dist-packages/motioneye/config.py", line 533, in get_camera
    _get_additional_config(camera_config, camera_id=camera_id)
  File "/usr/local/lib/python3.9/dist-packages/motioneye/config.py", line 2367, in _get_additional_config
    (sections, configs) = get_additional_structure(camera=bool(camera_id))
  File "/usr/local/lib/python3.9/dist-packages/motioneye/config.py", line 2338, in get_additional_structure
    result = func()
  File "/usr/local/lib/python3.9/dist-packages/motioneye/controls/tzctl.py", line 127, in timeZone
    import pytz
```

MichaIng, 2023-09-15:
Ah I see. Please try to comment this setting out: https://github.com/motioneye-project/motioneye/blob/ad21146/motioneye/extra/motioneye.conf.sample#L68C2-L68C32

Actually this cannot work anyway unless motionEye runs as root, while it does run as `motion` user since v0.43: https://github.com/motioneye-project/motioneye/blob/dev/docker/entrypoint.sh

There are some other settings which similarly require additional dependencies as well as elevated permissions, like the [SMB shares](https://github.com/motioneye-project/motioneye/blob/ad21146/motioneye/extra/motioneye.conf.sample#L56) feature, and shutdown/reboot from from GUI. IMO it is without doubt the only correct thing to run a web facing application like this as dedicated non-root service user with minimal permissions, but we need to document this properly and either find some way to enable these additional features without root permissions, via tailored `sudoers` for example.

However, in case of the time zone, IMO we could simply remove this all together. A correct time zone should be set for the whole system, in case of containers at host level, anyway. We should not aim to make a system administration tool from motionEye.

Bluscream, 2023-09-15:
Everything works now. Thank you :)

MichaIng, 2023-09-16:
Actually a good idea. Could you please create a new issue/request for this?