motioneye-project/motioneye #1898: attempting to set motioneyes os to record to an attached USB drive and I cannot find the device path to enter in motioneyes

swaziguy, 2020-08-24:


Johnny-quattro, 2020-10-20:
Check to see if the disk is recognised:

**_sudo dmesg -T_**
(possibly appears as /dev/sda)

**_lsusb_**

If the disk is indeed mounted, look in **_/media/..._**

Here's an example from my own Pi, with a Samsung external HDD:

**_[Tue Oct 20 11:09:48 2020] sd 0:0:0:0: [sda] 312581808 512-byte logical blocks: (160 GB/149 GiB)
[Tue Oct 20 11:09:48 2020] sd 0:0:0:0: [sda] Write Protect is off
[Tue Oct 20 11:09:48 2020] sd 0:0:0:0: [sda] Mode Sense: 28 00 00 00
[Tue Oct 20 11:09:48 2020] sd 0:0:0:0: [sda] No Caching mode page found
[Tue Oct 20 11:09:48 2020] sd 0:0:0:0: [sda] Assuming drive cache: write through
[Tue Oct 20 11:09:48 2020] sd 0:0:0:0: Attached scsi generic sg0 type 0
[Tue Oct 20 11:09:48 2020]  sda: sda1
[Tue Oct 20 11:09:48 2020] sd 0:0:0:0: [sda] Attached SCSI disk_**

starbasessd, 2020-11-04:
Closing, this is a motionEyeOS issue. If you need further assistance ask at
https://github.com/ccrisan/motioneyeos/issues

closed 2020-11-04 by starbasessd