motioneye-project/motioneye #1901: MMAL Camera Not Picking Up When Adding Cam - Raspberry Pi 4 8Gb

Tobinuk, 2020-08-28:
Hi,

I have 2 SD cards. One installed with MotionEyeOS & the other has Raspbian installed....

The MotionEyeOS SD has been flashed and seem to install and boot with the wifi supplicant files all booting done!

When I access MotionEye through the browser and IP address I can login as admin and get asked to add a cam. My cam is a nightvision camera purchased on ebay. The ribbon type. However it does not display under MMAL cameras when adding.

There are "No cameras"??

Why is this not picking up??

I ran a sudo command (forget now) which stated no cameras supported & no cameras installed?!

I shutdown MotionEyeOS and switched SD cards. 

When my Raspbian booted up I configured the camera, rebooted and I was able to take video and shots, then save to Rasp desktop....... So I know the camera works and that I have installed/fitted correct.

I then switched SD cards again...... Going back to MotionEyeOS, booted up and returned to my browser after successful boot.
Still no camera detected when trying to add MMAL cam :(

Does anyone have any ideas please?? I am very new to this. However I dont like to be beaten :)

Thanks for ready. Hope we can sort this!!

Johnny-quattro, 2020-10-20:
Check to see if the camera is connected properly and supported first:
**_/usr/bin/vcgencmd get_camera_**

Mine reports the following:
**_supported=1 detected=1_**

Secondly, I had a problem adding the camera and I eventually deduced that the motion daemon was being started and grabbing control of the camera. I disabled motion from starting by editing the file **_/etc/default/motion_**.

Hope this helps.

starbasessd, 2020-11-04:
Closing, this is a motionEyeOS issue. If you need further assistance ask at
https://github.com/ccrisan/motioneyeos/issues

closed 2020-11-04 by starbasessd