motioneye-project/motioneye #1907: arch linux pycurl undefined symbol

kmsgli, 2020-09-07:
Just updated my system and I would imagine something is up with the latest python update I took.  Motioneye is giving me this error when it tries to start
`● motioneye.service - motionEye Server
     Loaded: loaded (/usr/lib/systemd/system/motioneye.service; enabled; vendor preset: disabled)
     Active: failed (Result: exit-code) since Mon 2020-09-07 10:25:59 EDT; 1min 19s ago
    Process: 411 ExecStart=/usr/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=1/FAILURE)
   Main PID: 411 (code=exited, status=1/FAILURE)

Sep 07 10:25:59 archdesktop meyectl[411]:     import motionctl
Sep 07 10:25:59 archdesktop meyectl[411]:   File "/usr/lib/python2.7/site-packages/motioneye/motionctl.py", line 28, in <module>
Sep 07 10:25:59 archdesktop meyectl[411]:     import mediafiles
Sep 07 10:25:59 archdesktop meyectl[411]:   File "/usr/lib/python2.7/site-packages/motioneye/mediafiles.py", line 41, in <module>
Sep 07 10:25:59 archdesktop meyectl[411]:     import uploadservices
Sep 07 10:25:59 archdesktop meyectl[411]:   File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 27, in <module>
Sep 07 10:25:59 archdesktop meyectl[411]:     import pycurl
Sep 07 10:25:59 archdesktop meyectl[411]: ImportError: /usr/lib/python2.7/site-packages/pycurl.so: undefined symbol: PyUnicode_DecodeLocale
Sep 07 10:25:59 archdesktop systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Sep 07 10:25:59 archdesktop systemd[1]: motioneye.service: Failed with result 'exit-code'.
`

Any help would be appreciated as my cameras are down right now.

zsmbrvr, 2020-09-08:
I've got the same issue.

kmsgli, 2020-09-10:
So whats the solution here? downgrade pycurl will probably work but that cant be the solution, it seems it will no longer be supported b python 2 so what is the work around here?

kmsgli, 2020-09-30:
I dont see much added to any of the bug reports, is there anyway around this for motioneye?

c3ypt1c, 2020-10-03:
> So whats the solution here? downgrade python2-pycurl to version 7.43.0.4-1 has fixed motioneye but that is not a long term solution is it?

That seems to work at least temporarily. Worked for me on Arch.

JSkier21, 2020-10-05:
Long term solution would be to switch to python3 library of pycurl, IMO.

I have updated AUR package for motioneye and also added at temporary pycurl package in AUR for now.

kmsgli, 2020-10-06:
> Long term solution would be to switch to python3 library of pycurl, IMO.
> 
> I have updated AUR package for motioneye and also added at temporary pycurl package in AUR for now.

Thanks for the update, I had to uninstall and reinstall motioneye for it to install properly. Pacman was not happy removing python2-pycurl without manual intervention.

This is a good fix for now. Thanks Jskier21

starbasessd, 2020-11-05:
Several distros have already removed a bunch of python 2.7 dependencies for pip2, and such. I am not looking forward to Jan 2021 the final EOL date for 2.7...

closed 2020-11-05 by starbasessd