motioneye-project/motioneye #1910: Motioneyeos keeps on rebooting

nandomendoza, 2020-09-10:
I install a fresh copy of motioneyeos-raspberry-20200606.img on my raspberry pi zero w, I config my wpa_supplicant.conf file and i can see the meye ip address when i do a ip scan but when it gets to the line where it 
says starting motioneye it gets stuck for a min or 2 then it rebooting the system over. do you have an idea whats wrong
and how can i fix it.

nickmartin1ee7, 2020-09-16:
I am so experiencing this issue

Nando52, 2020-09-17:
I solved the problem, click on —> wiki, click on Supported Devices, download the file for you device, flash your micro sd card, this should fix the problem.

starbasessd, 2020-11-04:
Closing, this is a motionEyeOS issue. If you need further assistance ask at 
https://github.com/ccrisan/motioneyeos/issues

closed 2020-11-04 by starbasessd