motioneye-project/motioneye #1911: Frame change threshold and Show frame changes?

DevilWAH, 2020-09-11:
Why is the frame threshold in % while the Show frame changes is in actuly frames. Would it not be much better to use the same format for both? This would make setting the threshold much easier. we could simple enable the show changes, note the value and set the threshold to slightly higher. 

But when i have an average peek on a quiet image of 15 for the shown changes, what % does that mean in terms of a threshold?

zagrim, 2020-09-12:
Frame changes unit is actually in pixels, not frames, but yeah, the unit inconsistency makes numerical tuning a bit difficult. Then again, I do see why the frame change threshold is in percent, because it is quite impossible (at least for most people) to have an idea whether the threshold converts to 1/8 of the frame area, 1/30 of it, or half of it. If one is willing to tune just by the pixel change numbers, then that would be totally fine I guess.

However, there is a proposed solution for this in #1441 which would work both ways.

zagrim, 2020-09-12:
I have found out that unless the environment is totally controlled (no extra randomly moving objects in the frame) the only way to get a decent motion detection setup is run the system, observe the results and adjust the threshold not by the values (because there's no single "good" value) and then run it again, observe, adjust... But since I have no experience on tuning in a controlled setup, my view on the importance of the topic here is likely biased.

starbasessd, 2020-11-04:
I use Despeckle ON, Show Frame Changes ON, and watch where it sees changes. If it is something like leaves moving or grass moving I consider masking out that area. Despeckle turns off sparklies from water reflections for me. Due to the sensitivity, though, YMMV
Do you need further assistance?

JamBalaya56562, 2026-11-10:
Closing as a duplicate of #1441, which asks for the same: showing the frame change threshold in pixels, as Show Frame Changes does, next to the percentage.

closed 2026-11-10 by JamBalaya56562