motioneye-project/motioneye #1913: simple MJPEG camera username/password is not stored

cesarandreslopez, 2020-09-26:
On MotionEye GUI, if you add a Simple MJPEG camera it will allow you to set a username and password for the mjpeg URL. As long as the session is open, the stream will work well, but if the browser session is closed, and the reopen the stream will not show.

Checking the camera-n.conf file generated you will get something as follows:

```
# @id 1
# @enabled on
# @proto mjpeg
# @url http://xx.xx.xx.xx/mjpg/video.mjpg

camera_name Camera1
```

But no username/password seems to be stored.

On the browser's console, the URL above will show a 401.

codelover96, 2020-10-27:
I am not very familiar with how the simple mjpeg camera works. My guess is that, the username and pass are stored somewhere else for safety, once you setup the camera for the first time. 
Maybe I'm wrong.

starbasessd, 2020-11-04:
I am running a setup on an I3 Lenovo TinyPC, Ubuntu 20.04, motion 4.3.1, motionEye 0.42.1, without that issue. Can you post your setup, please?

closed 2021-03-05 by starbasessd