motioneye-project/motioneye #1915: Why is motionEye loading just "jpeg images" as the feed? Why not using mpeg stream?

developerfromjokela, 2020-09-27:
Hello!
I'm just thinking, motionEye currently loads just constantly a snapshot of the camera image. 
Why wouldn't we use mpeg streams? Why the extra code to constantly fetch images?

developerfromjokela, 2021-03-05:
Hmm what is this? Ignoring user feedback?

starbasessd, 2021-03-05:
No, trying to clean up OLD issues. Was going to go back and re-open labeled issues like bugs, questions, enhancement a feature requests. There hadn't been any feedback or comments since I labeled it a question back in November. Trying to make it easier for the team to find active issues, and stuff that hadn't been resolved by newer updates.

developerfromjokela, 2021-03-05:
Ok

JamBalaya56562, 2026-11-10:
Closing with the answer: the UI fetches `picture/<id>/current/` frames through motionEye itself, so the picture comes over motionEye's one port with its login and also works for remote cameras and behind a proxy, where Motion's own MJPEG ports are not reachable. The MJPEG stream is still there under Video Streaming for other players.

closed 2026-11-10 by JamBalaya56562