motioneye-project/motioneye #1916: Issues with motion notifications URL with "%20" and Show Frame Changes dropping to 0.

YuuichiLV, 2020-09-28:
Greetings,

It seems when I try to use either cURL or WebHook if the URL has "%20" in it, it doesn't work. Also the show frame changes output drops to 0 every few seconds for about 2 seconds each, is affecting multiple cameras. I have also tried this on my Debian 10 server, not raspberry pi, and getting the same issue on both problems. The cameras are RTSP set low to 352x288. CPU load is like 10% for 5 cameras on Pi 4.

I am using a fresh install following the Raspberry Pi install instructions. Didn't want to use the Motion Eyes OS because it shows to be using Stretch and not Buster, shows using older kernel.

I am using Homebridge Camera FFmpeg which accepts webhooks to activate motion censor on the generic camera. Example URL is "http://10.0.0.1:8877/motion?Front%20Yard". This command works correctly when can from CLI with curl -k. This is on the same Pi.

If I change the name of the camera in Homebridge Camera FFmpeg to the following example "http://10.0.0.1:8877/motion?FrontYard", it works. Somehow the output to cURL isn't correct. I can't find any log errors and doing a "journalctl -u motioneye > motioneye.log" doesn't show any errors or problems found.

Journalctl output : 
Sep 28 15:21:44 Pi-HB-Cameras motion[15489]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2 Started
Sep 28 15:21:44 Pi-HB-Cameras motion[15489]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Sep 28 15:21:44 Pi-HB-Cameras motion[15489]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Sep 28 15:22:19 Pi-HB-Cameras motion[15871]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Sep 28 15:22:19 Pi-HB-Cameras motion[15871]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
Sep 28 15:22:19 Pi-HB-Cameras motion[15871]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-2.conf
Sep 28 15:22:19 Pi-HB-Cameras motion[15871]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-3.conf
Sep 28 15:22:19 Pi-HB-Cameras motion[15871]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-4.conf
Sep 28 15:22:19 Pi-HB-Cameras motion[15871]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-5.conf
Sep 28 15:22:19 Pi-HB-Cameras motion[15871]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
Sep 28 15:22:19 Pi-HB-Cameras motion[15871]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2 Started
Sep 28 15:22:19 Pi-HB-Cameras motion[15871]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Sep 28 15:22:19 Pi-HB-Cameras motion[15871]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Sep 28 15:43:48 Pi-HB-Cameras meyectl[9242]:  WARNING: 403 GET /login/?_=1601333028416&_username=_&_signature=fc7459430c36a5f5dbeb891b22902b106f6062fc (10.0.1.20) 1.47ms

starbasessd, 2020-11-04:
It has known issues using spaces (%20) in filenames/paths. 
Have you tried motioneyeOS for 20200606, or dev20201926?

closed 2021-03-05 by starbasessd

hajuegei, 2023-08-21:
Hi all,
is the  problem solved?

zagrim, 2023-08-22:
Do you have that problem? Have you tried with the [current dev branch](https://github.com/motioneye-project/motioneye/tree/dev) (pending official release), or are you using MotionEyeOS (in which case the most recent still would be the ones mentioned in the above comment)? I doubt that there's a fix in the old Python 2 codebase (that is, the official releases) for this, but the dev branch (Python 3) just might work better - still no guarantees of any kind.

hajuegei, 2023-08-22:
I have this problem by controlling esp32 devices under tasmota with webhook. tasmota will receive a webstring like http://<ip>/cm?cmnd=power%20<On/Off>. Motioneye is possible the best solution for all raspis. Thats why I will use the branch and hope it works.

hajuegei, 2023-08-22:
Many thanks,

I have found a solution — DietPi with motionEye for all raspberries

zagrim, 2023-09-27:
Just being curious @hajuegei : Did using the current dev branch code resolve your issue, or did you find some other resolution?