motioneye-project/motioneye #1917: Motioneye not running on RaspberryPi after minor edits to html templates and docker build

edsase, 2020-09-29:
I followed the build instructions for building Motioneye from a Docker image. After cloning the Github repo, I did some edits to the html templates before running docker build command. However, running the container image does not work as expected. The usual initial login modal doesn't pop up but rather, I get directly to the main page. The menu on the main page is not working and so there is nothing I could do on that page. On the other hand, when I clone the repo and build the image without first editing the html templates, I don't get any issues. Could it be that for the docker install the html templates should not be edited? Thanks already for your responses!

yurnov, 2020-10-13:
It depends on what the change you did.

edsase, 2020-10-14:
I compiled with the latest Docker image and the issue did not appear. Thanks @ccrisan

closed 2020-10-14 by edsase