motioneye-project/motioneye #1919: Using movie_extpipe_use/movie_extpipe

gusarg81, 2020-10-07:
Hi,

I have a doub using these parameters (my intention is to use movie_extpipe so I can encode using qsv. Ex: h264_qsv and is the only way that Motion supports this particular codec to be used).

So, when using this paramenter, I must disable "Movies" in the frontend?

Thanks.

mtmongoose, 2020-10-08:
I'm also curious about the use of Motion options movie_extpipe_use and movie_extpipe with MotionEye since I'd like to capture audio and use a video codec that is not available from within the MotionEye dropdown. Currently while using these settings I end up with two movies per motion-triggered event. I tried setting the movie quality to 0 within MotionEye but still ended up with two movies per motion detection event. It would be great if MotionEye would recognize that the option movie_extpipe_use was on and not also try to record a movie. Also, my understanding of these two options can lead to better performance since the recording of the movie can be offloaded to a different thread.

Thanks!

Your MotionEye project is awesome and I've tried just about every software NVR/VMS available except for Blue Iris. If Blue Iris ran on Linux or stably within a container without WINE I might use it.

gusarg81, 2020-10-08:
@mtmongoose can you share how you used motion_extpipe? Because in my settings it says that can't connect to pipe.

mtmongoose, 2020-10-08:
Sure, below is how I had one of my camera's config file set with the two
necessary options. I have the lines commented out currently.

#movie_extpipe_use on
#movie_extpipe ffmpeg -y -hwaccel auto -f rawvideo -pix_fmt yuv420p
-video_size %wx%h -framerate %fps -i pipe:0 -c:v libx264 -preset medium
-c:a audio -f mp4 %f.mp4

gusarg81, 2020-10-08:
Thanks!

I am using:

> movie_extpipe ffmpeg -y -init_hw_device qsv=hw -filter_hw_device hw -f rawvideo -pix_fmt yuv420p -video_size %wx%h -framerate %fps -i pipe:0 -vf hwupload=extra_hw_frames=64,format=qsv -c:v h264_qsv -c:a audio %f.mp4

But it gives me: Broken pipe :(

mtmongoose, 2020-10-08:
Do you get an error if you use my configuration for movie_extpipe?

gusarg81, 2020-10-08:
Well, is almost the same, just I use qsv encoder for hw accel on Intel and I don't think that is the problem by "Broken pipe", but I could be wrong.

mtmongoose, 2020-10-08:
I'm running MotionEye in a Docker container and when I run "FFmpeg -codecs
| grep qsv" in the container console, I don't find the codec h264_qsv as
available. Could that be the problem? I'm not an expert in all of this, so
I could be wrong.

gusarg81, 2020-10-08:
Nope, that is no the problem. QSV stands for QuickSync which is used for Intel for hardware accel. So, is not enabled and not installed by default (is that why you don't see it).

Anyways, tryed with vaapi as well and does not work either, but all this is another subject for Motion project itself which I've already asked there (since motioneye does nothing about decoding/encoding).

gusarg81, 2020-10-08:
For the original question of this issue, after added movie_extpipe I've just disabled Movies in the frontend and I can also still play the stored movies (by movie_extpipe) in the frontend as well.

mtmongoose, 2020-10-09:
@gusarg81 Does recording get triggered by motion or are you even using the motion record option?

mtmongoose, 2020-10-09:
@gusarg81 I checked with ffmpeg -hwaccels and it does not look like ffmpeg is compiled with QuickSync (qsv) support.

Hardware acceleration methods:
vdpau
vaapi
drm

gusarg81, 2020-10-09:
> @gusarg81 Does recording get triggered by motion or are you even using the motion record option?

Yes, it does trigger by motion (which you can check this in motion logs)

gusarg81, 2020-10-09:
> @gusarg81 I checked with ffmpeg -hwaccels and it does not look like ffmpeg is compiled with QuickSync (qsv) support.
> 
> Hardware acceleration methods:
> vdpau
> vaapi
> drm

You misunderstood. What I meant is that your FFmpeg does not have qsv enabled (and that because is not enabled/compiled with qsv support by default). Mine does have enabled because I've compiled it by hand, enabling qsv support.

closed 2021-03-05 by starbasessd