motioneye-project/motioneye #1922: unable to compile on Rpi4 4gb 64bit

puccaso, 2020-10-11:
Hello!

I was getting an error while compiling from the git clone. initially, I didn't have the interface/mmal/mmal.h in the source, so I downloaded and copied it, however now I am getting an LD issue.. 

`root@flux:/home/pi/Desktop/a/motion# make
make  all-recursive
make[1]: Entering directory '/home/pi/Desktop/a/motion'
Making all in src
make[2]: Entering directory '/home/pi/Desktop/a/motion/src'
/usr/bin/distcc -D_THREAD_SAFE  -I/usr/include/p11-kit-1  -Irasppicam  -I/usr/include/aarch64-linux-gnu  -I/usr/include/mariadb  -g -O2   -o motion motion.o logger.o conf.o draw.o jpegutils.o video_loopback.o video_v4l2.o video_common.o video_bktr.o netcam.o netcam_http.o netcam_ftp.o netcam_jpeg.o netcam_wget.o netcam_rtsp.o track.o alg.o event.o picture.o rotate.o translate.o md5.o ffmpeg.o webu.o webu_html.o webu_stream.o webu_text.o mmalcam.o raspicam/RaspiCamControl.o raspicam/RaspiCLI.o  -pthread   -ljpeg  -lmicrohttpd  -lmmal -lmmal_core -lmmal_util -lmmal_vc_client -lbcm_host -lvcsm -lvcos  -lvchostif  -lavutil -lavformat -lavcodec -lswscale -lavdevice  -lmariadb -lpthread -lz -ldl -lm -lgnutls 
/usr/bin/ld: cannot find -lmmal
/usr/bin/ld: cannot find -lmmal_core
/usr/bin/ld: cannot find -lmmal_util
/usr/bin/ld: cannot find -lmmal_vc_client
/usr/bin/ld: cannot find -lvcsm
collect2: error: ld returned 1 exit status
distcc[30071] ERROR: compile (null) on localhost failed
distcc[30063] ERROR: compile (null) on localhost failed
make[2]: *** [Makefile:422: motion] Error 1
make[2]: Leaving directory '/home/pi/Desktop/a/motion/src'
make[1]: *** [Makefile:554: all-recursive] Error 1
make[1]: Leaving directory '/home/pi/Desktop/a/motion'
make: *** [Makefile:398: all] Error 2
root@flux:/home/pi/Desktop/a/motion# 
`

I'm missing something, I know I am but I just cannot figure out what. 

I'd be thankful for any help!

puc.

zagrim, 2020-10-12:
You were not just missing the headers, you're also missing the libraries (libmmal, libmmal_core etc). I think you should install the "dev" versions of whatever "mmal" packages there are (it's not clear to me what distro you are using).

starbasessd, 2020-11-04:
Did that resolve your issue?

closed 2021-03-05 by starbasessd