motioneye-project/motioneye #1923: Create album per day when uploading to photos

smhgit, 2020-10-16:
Add the option to create album per day when uploading to google photos. The album name will be based on prefix concatenate with date

codelover96, 2020-10-27:
I recommend marking this issue as "enhancement" or "feature-request"

starbasessd, 2020-11-04:
For Still Images, File Name, I use : %Y-%m-%d/%H-%M-%S which gives me Year-Month-Day/and timestamp for the file name.

closed 2020-11-04 by starbasessd

smhgit, 2020-11-04:
Why did you close the issue? Won't you support ALBUM per day?

starbasessd, 2020-11-04:
Album per day is done by setting Filename under Movies or Still images to:
%Y-%m-%d/%H-%M-%S
This gives you a folder for each day, and the time stamp of each file as
the filename.
So, wherever you have you files saving under File Storage, you will have a
folder (like today would be:
2020-11-04/
With filenames under it like 14-23-07.jpg  or 14-23-07.mp4

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

smhgit, 2020-11-05:
@starbasessd yes, it works when you upload to google drive (I already have this), but not for Photos as far as I checked. I will double check it, but I think for phots it doesn't create an album.