motioneye-project/motioneye #1925: How to update config using command line - motioneyeOS

ZKAW, 2020-10-16:
Hello everyone.

I'm running motioneyeOS and I need to know how to apply the changes made to te .conf file using command line instead of the web interface.
I'm looking to do a script to adjust the brightness for day/night cycle.

Thank you, and if you need any more information tell me.

Johnny-quattro, 2020-10-20:
Please be aware that this is the github repo for motioneye, not motioneyeOS

On the subject of day/night, I've experimented with the 'sunwait' utility and had good results, so that option might be worth investigating?

I commented about the syntax required when setting up a cron task to restart at certain times here:
https://github.com/risacher/sunwait/issues/2

ZKAW, 2020-10-20:
> Please be aware that this is the github repo for motioneye, not motioneyeOS
>
> On the subject of day/night, I've experimented with the 'sunwait' utility and had good results, so that option might be [...]

Okay thanks for the reply. I didn't know about the existence of a different repo for motioneyeos. And what is "sunwait"? Never eared about it. Can you explain me how it's working?

Johnny-quattro, 2020-10-20:
I left a clickable link in my last post detailing the use of 'sunwait' in conjunction with a cron task.

The main page for that repo is here: **_https://github.com/risacher/sunwait_**
Usage: **_https://github.com/risacher/sunwait/blob/master/USAGE.txt_**

Johnny-quattro, 2020-10-20:
here's an example of how I use it in a crontab entry:

**_# set restart of motioneye daemon to clear the over-exposed/whiteout camera image (only happens on Pi3)
@reboot /home/pi/scripts/sunwait wait rise offset +0:25:00 51.360258N -0.147890E; sudo service motioneye restart_**

closed 2021-03-05 by starbasessd