motioneye-project/motioneye #1928: Using WinSCP to transfer files to network drive

Bromley47, 2020-10-19:
I am using WinSCP to transfer media from the Pi Camera SD card to an External hard drive attached to a Windows 10 PC. This works fine with Rpi Web Cam program and the files are found in /<root>/var/www/media.

However in Motioneyeos they appear to be  in /<root>/data/20-10-19 being the date the media was saved.

I can point WinSCP at the the var/www/media*.* which transfers all the media to the hard drive but will this work with the date folder and how do I get the dated folder to transfer?

Any help appreciated.

codelover96, 2020-10-27:
What protocol are you using? FTP?

Bromley47, 2020-10-31:
Using SFTP

starbasessd, 2020-11-03:
I do a pscp (Win10 putty scp app)
pscp -r user@pi:/home/pi/sdcard/ \Users\me\Downloads\ [enter]
Change your path to root of folder you want to get from, and where you want to store.
Be sure to include the trailing '/' and '\' stop the path at the folder where the date folder is.

closed 2021-03-05 by starbasessd