motioneye-project/motioneye #1930: RTMP stream: unable to open video device

jarneg, 2020-10-22:
Hello, i'm streaming RTMP via my OSMO action action camera.
as camera device i'm using;
rtmp://192.168.1.123:1935/live/tester

I'm able to play this via VLC but if i'm adding this in MotionEye i get a grey screen and "unable to open video device".

If i'm streaming via OBS to this URL it's working.
I can input the stream in OBS rtmp://192.168.1.123:1935/live/tester and the restream it to something like
rtmp://192.168.1.123:1935/live/tester1
This rtmp is visible in motionEye.

Any idea what is wrong with the direct stream?

Installed motion via:
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian

CRAWLER-V, 2020-10-31:
i'm join the same issue but with "RTMPS" stream from 2 google nest cam outdoor.
Motioneyeos installed on raspberry pi 4 (4gb ram)

starbasessd, 2020-11-03:
Nest uses an encrypted data stream not supported by motioneye.

closed 2021-03-05 by starbasessd