motioneye-project/motioneye #1934: 2 XBox 360 cameras connected to Pi Model B only one showing picture

slickochet, 2020-10-26:
[image]

When connecting 2 360 USB cameras to my motioneye it only seems to get a picture from one.  This seems to be some setting getting messed up because of them being the same model but I can't figure out which.  I think this because when I connect 2 cameras of different models they both work fine.

I get the following error in motion on startup:

[2:ml2:Camera2] [ERR] [VID] v4l2_input_select: Unable to query input 0. VIDIOC_ENUMINPUT, if you use a WEBCAM change input value in conf by -1: Inappropriate ioctl for device
[2:ml2:Camera2] [ERR] [VID] vid_start: V4L2 device failed to open
[2:ml2:Camera2] [WRN] [ALL] motion_init: Could not fetch initial image from camera 
[2:ml2:Camera2] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[1:ml1:Camera1] [WRN] [VID] v4l2_pixfmt_set: Adjusting resolution from 320x200 to 320x240.
[2:ml2:Camera2] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera

URLs that motioneyeos picked up for them were the following:

/dev/v4l/by-id/usb-Microsoft_Video_Camera_000F330238624704-video-index1
/dev/v4l/by-id/usb-Microsoft_Video_Camera_000F330238624704-video-index0

I tried manually changing them to /dev/video0 and /dev/video1 but with the same result.

Does anyone have any suggestions about how I can get these 2 identical cameras working at the same time?

Thanks in advance

slick

starbasessd, 2020-11-03:
The system cannot differentiate 2 usb cameras with the same vendor:product ID reliably.

slickochet, 2020-11-03:
Yeah, I suspected as much.  I have switched to using 2 different cameras for now thanks!

closed 2020-11-03 by slickochet