motioneye-project/motioneye #1935: ESP32-CAM

lodv, 2020-10-26:
Hi,

I installed MotionEye in Home Assistant.
I want to add my ESP32-CAM, but that's not possible.
I get the message that the camera is not a supported network camera.
The camera has an ip address and I can open the stream in a browser.
Everywhere I see tutorials where they add ESP32-CAM into Home Assistant, so I'm wondering if you know what's going on.
Camera and Home Assistant (running on a Raspberry Pi) are attached to the same network.
Thank you for your help.
Yvonne

slash153, 2020-10-29:
assuming you use the example streaming server, you need to use "http://192.xxx.xxx.xxx:81/stream" (without quotes) to add the video stream into motioneye/motioneyeos. Please note it defaults to lowest settings, if you need a better resolution, you need to change it in the esp32-cam administration and it resets to default after reboot/power loss.

lodv, 2020-10-29:
Thank you so much. That is the right answer.

Kind regards,
Yvonne

closed 2020-11-03 by starbasessd