motioneye-project/motioneye #1938: Getting an Auto Focus Pi Camera to auto focus on Motioneye

pachimzar, 2020-10-29:
I am trying to configure Motioneye to take advantage of the auto focus feature of my ArduCam Autofocus Camera  (B0176).  I tried to use "on_event_start  in motion.conf.  I wrote a python script and imbedded it into a bash script. No success. 

Can Motioneye or Motion be configured to do auto focus? If so, can you point me to a resource for help.

Thanks,
pachimzar

camaro4life18, 2021-02-11:
I'm planning to do the same, should have my camera here tomorrow. I was thinking of seeing the focus the same way, but after reading your comment and the how to on ardurcams site, once your script stops, the focus resets back to normal. On_event_start would just trigger the script to run, then stop. It would need to run all the time, or at least in a loop.

I do have an idea, kind of a hack similar to the audio hack, but you could put the script into a loop and start it with a pm2 start command, then in the post file writing run pm2 stop.

JamBalaya56562, 2026-11-10:
Closing, as this got no answer: Motion has no focus control, so an autofocus module like the Arducam B0176 has to be driven by Arducam's own focus tools outside motionEye, for example from Run A Command or an action button script.

closed 2026-11-10 by JamBalaya56562