motioneye-project/motioneye #1939: Pi 4b Cam not detected

DPBiff, 2020-10-31:
I have turned my brain to mush trying to get this sorted. 

I own a RPi 4b, 8Gram, 129Gb, SD SanDisk, Arducam 1/4 Inch 5 Megapixels Sensor  Camera Module with Flex Cable for Raspberry Pi Model A/B/B+, Pi 2 and Raspberry Pi 3, 3B+, 4 
OS: Buster
cat /etc/os-release
     PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
     NAME="Raspbian GNU/Linux"
     VERSION_ID="10"
     VERSION="10 (buster)"
     VERSION_CODENAME=buster
     ID=raspbian
     ID_LIKE=debian

I have:
- sudo apt update
-sudo apt upgrade
- Installed motioneye according to the instructions
- Installed motioneye via 'pip'
- Installed / pulled / Docker motioneye container
- Installed motioneyeOS on 64Gb SD - no joy couldn't get the os image to boot

Installing motioneye myself results in a failure to create a web interface
Installing Docker motioneye results in a web interface that can't find a cam

Results of motion -h:
motion Version 4.2.2, Copyright 2000-2019 Jeroen Vreeken/Folkert van Heusden/Kenneth Lavrsen/Motion-Project maintainers

Raspi-config: cam enabled

Raspistill captures .jpg

Raspivid captures  video

Motion captures video

ls /dev/video*  returns: 
         ls /dev/video01

vcgencmd get_camera returns:
        supported=1 detected=1

network: wifi and / or ethernet

sudo rpi-eeprom-update
- > BCM2711 detected
- > VL805 firmware in bootloader EEPROM
- > BOOTLOADER: up-to-date
- > CURRENT: Thu 03 Sep 2020 12:11:43 PM UTC (1599135103)
- >  LATEST: Thu 03 Sep 2020 12:11:43 PM UTC (1599135103)
- >  FW DIR: /lib/firmware/raspberrypi/bootloader/critical
- > VL805: up-to-date
- > CURRENT: 000138a1
- >  LATEST: 000138a1

I've run out of ideas -- help? Please?

DPBiff, 2020-11-04:
Ok,  always grateful for the help & suggestions... 
I have purchased a new 64Gb  sdcard, installed a fresh install of Raspbian, and a fresh install of Motioneye. 
                         It works. It works just fine.
I have concluded that there must be some other piece of software on the original install that's grabbing control of the camera. The only other significant piece of s/w I have going is Openplotter.  OP includes: 

- I2c
- CAN bus
-  Signal K
- SDR VHF
- AIS
- ADS-B
- CQRX
- DAV
- DBTV
Does that ring any bells for anyone?

starbasessd, 2020-11-04:
Did you enable the camera in raspi-config?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

DPBiff, 2020-11-05:
> Did you enable the camera in raspi-config?
As noted in the OP, yes.

I have, in the mean time, figured it out.   It was an install issue.

- removed all components of OpenPlotter - no change
- Ran "systemctl status motioneye" - this returned errors, one error at a time
The first error was a motioneye.conf file missing in a certain directory. I copied the file there from /pi/motioneye
The second error was a missing 'log' file directory. I created the dir 
The third error was a missing 'run' directory. I created the dir
The fourth error was Errno 98 port already in use. I checked it wasn't but on a different post I discovered that this error would throw if 'motion' was auto-started at boot. I set motion to start at boot and ... voila!  Motioneye now 'sees' my cam. Working as expected.
I reinstalled Openplotter components with no further issues.

Before anyone suggests that I didn't run pip as sudo  - - - - I did. Something went hinky in the install but addressing the errors thrown when running "systemctl status motioneye" did  the trick.

closed 2020-11-05 by DPBiff