motioneye-project/motioneye #1940: Docker build fails with "E: Could not configure 'libc6:amd64'

pthorel, 2020-11-01:
I'm following docker instructions for custom user (which need to build a custom docker image).
Building fails with following output :

```
Processing triggers for libc-bin (2.31-4) ...
E: Could not configure 'libc6:amd64'.
E: Could not perform immediate configuration on 'libnss-nis:amd64'. Please see man 5 apt.conf under APT::Immediate-Configure for details. (2)
The command '/bin/sh -c echo "deb http://snapshot.debian.org/archive/debian/20201027 sid main contrib non-free" >>/etc/apt/sources.list &&     apt-get update &&     DEBIAN_FRONTEND="noninteractive" apt-get -t stable --yes --option Dpkg::Options::="--force-confnew" --no-install-recommends install       curl       ffmpeg       libmicrohttpd12       libpq5       lsb-release       mosquitto-clients       python-jinja2       python-pil       python-pip       python-pip-whl       python-pycurl       python-setuptools       python-tornado       python-tz       python-wheel       v4l-utils &&     DEBIAN_FRONTEND="noninteractive" apt-get -t sid --yes --option Dpkg::Options::="--force-confnew" --no-install-recommends install       motion       libmysqlclient20 &&     sed -i -e "s/^\(motion:[^:]*\):[0-9]*:[0-9]*:\(.*\)/\1:${RUN_UID}:${RUN_GID}:\2/" /etc/passwd &&     sed -i -e "s/^\(motion:[^:]*\):[0-9]*:\(.*\)/\1:${RUN_GID}:\2/" /etc/group &&     pip install /tmp/motioneye &&     rm -rf /tmp/motioneye &&     apt-get purge --yes python-setuptools python-wheel &&     apt-get autoremove --yes &&     apt-get --yes clean && rm -rf /var/lib/apt/lists/* && rm -f /var/cache/apt/*.bin' returned a non-zero code: 100
```

Steps to reproduce : follow docker installation instructions with custom UID/GID

I've tried reverting last Dockerfile commit in order to use other debian snap srces, but did not manage to fix it that way.
Then done some research : seems like a debian/docker bug which might be fixed externaly.
Found : 
 * https://github.com/docker/for-linux/issues/1131
 * https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=972552

I'm still trying to find a workaround or fix and will comment here if I find anything usefull

pthorel, 2020-11-01:
Reading above bugs report led me to the following :
* That's a known apt bug which is already fixed but not yet deployed to released packages
* Adding this options to apt-gt on sid does pretty much the same as the fix : `-o APT::Immediate-Configure=0`
* The current Dockerfile may be fine without any change just waiting a few days or weeks ?

Beeing impatient, the following changes did the trick for me : 
https://github.com/ccrisan/motioneye/commit/05102133a27c9d86b65920b738f3d534cac5a886

starbasessd, 2020-11-03:
Not enough info, forgot the docker environment. x86/x64/arm?

closed 2021-03-05 by starbasessd