motioneye-project/motioneye #1942: cannot run both cameras, each one on its own works fine

AntouanK, 2020-11-01:
Hi there

having a bit of a weird issue.
I have a small laptop running arch and docker in there.
I am running motioneye via a docker container. Nothing weird there, I did that with a raspberry pi for the past 3 months, no problem at all.

I pass all the devices to the container
```
docker run --name="motioneye" \
    -p 8765:8765 \
    --hostname="motioneye" \
    -v /etc/localtime:/etc/localtime:ro \
    -v /etc/motioneye:/etc/motioneye \
    -v /var/lib/motioneye:/var/lib/motioneye \
    --restart="always" \
    --detach=true \
    --device=/dev/video0 \
    --device=/dev/video1 \
    --device=/dev/video2 \
    --device=/dev/video3 \
    ccrisan/motioneye:master-amd64
```
( btw why does each camera create two devices? from what I understand 0 and 1 is the first camera, and 2 and 3 is the second )

Anyway.
What's weird is that now that I switched to the laptop, I have 2 usb cameras ( identical ) and one of them is not working when I switch the resolution high.
Each camera works fine on its own. Goes up to 1920x1080, no problem.

Let's say I add camera X.
Works fine. I change resolution to any acceptable setting, all good.
As soon as I activate the camera Y, I get the grey screen wit the "unable to open video device" message.
If I set camera Y to 320x240, I get image there.
If I set camera X to 320x240, and camera Y to 1920x1080, all good!
As soon as I turn camera X to a higher resolution, again the grey screen to one of them.

To summarize: 
each camera on it's own, works fine on all resolutions. But cannot have both cameras with high resolution, one has to be at 320x240!
What can be the issue? Any ideas?

Thank you.

starbasessd, 2020-11-03:
Can you set both to 640x480? 800x600? 1024x768?

AntouanK, 2020-11-03:
> Can you set both to 640x480? 800x600? 1024x768?

none of those work. :(
Always one camera is not working. 
If I disable one of them, the other one works fine on any of those resolutions.

I was thinking another solution.
I have another PC at home, and I wanted to see if I can use it as the master motioneye server.
Can I launch 1 container per camera, and expose just the stream with the "Fast Network Camera" feature? ( not sure if that's possible since I cannot find the option )

That way, I solve the 2 cameras issue, and the CPU issue also since that PC is beefier and can handle the 2 cameras with motion detection.

starbasessd, 2020-11-03:
Fast Network Camera only works with CSI PiCam cameras (connected with a
ribbon cable)
Also, there are some features not available in the Docker version.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

AntouanK, 2020-11-03:
ok. thank you for the help.

so if I want to just expose my cameras to a 3rd motioneye server, I have to use something else, not motioneye, correct?
any suggestion?

back to the original issue, is there anything else to try?

starbasessd, 2020-11-03:
Put the 2 cameras on 2 separate PIs, and then combine to one of the pis, or
a third pi, or to a third motionEye ithey will work fine.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

texasStronger, 2020-11-20:
I get the gray screen when there is not enough video memory.  I can run /boot/config.txt gpu_mem at 256 on a RPI 4 8GB, but can only run 128 on RPI 4 4GB. If I change to 256 on 4GB gray video screen every time.

closed 2021-03-05 by starbasessd