motioneye-project/motioneye #1944: Unable to Edit Camera Setting

vincentlee20, 2020-11-03:
I have installed the motion and motioneye (pip) following the instruction. I can add a camera and able to edit the setup before logoff. However, when logoff and logon again with admin, all the configuration is dimmed and cannot be edited. I need to remove the camera and add the camera again to change the setting.
- Raspberry Pi4 4Gb
- Raspbian

By the way, does anyone try to use docker for motioneye with Raspberry Pi4? I got CPU >100% (overall CPU ~25%), even I disable motion detection and not logging on the Web. So I switch to normal pip install now (CPU ~60^, overall CPU 1x%)

starbasessd, 2020-11-03:
It looks like Docker and top show the CPU usage as composite, instead of
average. On the Pi4, I get between 80-100% usage, which actually should be
divided by 4. To prove it, you can get 120% cpu usage, which would be 30%
per CPU and it adds them together. It's a bug, either in Docker or top.

As to the first issue, what is your base OS you are installing on? Which
version of RaspberryPiOS? 2020-08-26, or other? Where did you log in from?
The PI? oOr another computer?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

vincentlee20, 2020-11-03:
I have compared docker installation and normal installation, it is 100% vs 50%...dividing by 4 it shows 25% vs 1x%. I try to add GPU device in docker but with no luck

I try to login from my laptop, but I am using the admin to logon motioneye.
My OS info is as below
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster

Thank you

starbasessd, 2020-11-03:
Where/how did you  image it? Download an image from RPiFoundation? When?
What link?
Or are you running something like HaaS, or OpenHab, or OMV?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

vincentlee20, 2020-11-03:
I get the raspbian OS from RPiFoundation about 6months ago.

HaaS, do you mean home assistant? I have it installed in Raspbian too.When I know I can't install the motioneye as addon, I follow the step to install motioneye manually. And I logon motioneye webUI to do the configuration. No integrate between 2 yet, hopefully I can make it later :)

starbasessd, 2020-11-03:
Have you done any sudo apt update/upgrades recently?
There are conflicts with Home Assistant, OpenHab, OMV and others. You are
REALLY asking the Pi to do a lot between any 2 of them. Video processing
takes a lot on it's own...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

vincentlee20, 2020-11-03:
The problem is that, I cannot edit the configuration. So I am suspecting if I did something wrong in the access right setting.

I am planning to buy another Pi4 to split the workload, while I am testing the motioneye before investing. Do I need to stop the Home Assistant which usually uses <10% CPU ?

starbasessd, 2020-11-03:
Have you done a reboot, since you are not answering my other questions?
Whether or not to run any other package is beyond the scope of my support
here, others may jump in to help with that, or you can ask on the Home
Assistant support pages...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

vincentlee20, 2020-11-03:
I have done sudo apt update/upgrade, and reboot. 
To install motioneye in Raspbian, I follow every steps of below
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian

vincentlee20, 2020-11-03:
I find something but I am not sure if it is a possible root cause. 

I find the below from the log. Does my motioneye run in Python 2.7? However, when I install motioneye by pip, pip is running in python 3.7
/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py"

starbasessd, 2020-11-03:
You may need to follow the Install on Ubuntu 20.04 or Newer instructions.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

vincentlee20, 2020-11-04:
Does it mean I should follow Ubuntu 20.04 but not the raspbian installation guide?

Since it runs well in docker, I am trying if I can add GPU as a device in docker so I can reduce the CPU usage from the current 100% usage in a CPU.

starbasessd, 2020-11-04:
It's actually only 25% per CPU. It's adding them for some reason, instead
of averaging them across 4 CPUs. You can tell when the CPU usage exceeds
100%. Mine hits 110% occasionally.
If you are running on the latest RaspberryPiOS, I have found some of the
apps missing (like pip2) or you need to specify pip2 install xxx... because
python2 is not the default.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

vincentlee20, 2020-11-05:
You are correct that I use pip3 to install, so it may be the reason of the non-editable configuration.

I am also trying to explore the docker way, as a learning curve for docker, Linux, etc.

icurnow, 2020-11-08:
Same problem with greyed out settings having correctly entered admin credentials. This is after upgrade to current dev version using in system checker running on Rpi2

starbasessd, 2020-11-08:
There isn't a dev version in motionEye, only motionEyeOS.
Yo umay need to start from scratch on the motion/motionEye install, this
time use the Ubuntiu 20.04 and newer instructions.
Make sure you use your sudo -i or sudo su before attempting any
installation of the software.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

vincentlee20, 2020-11-09:
I try to run it in docker and keep it around 50-100% on 1 CPU (25%), but I have to set the frame rate as 2 per second. Since it is a surveillance camera, it is fine to have low frame rate to make it work :)

icurnow, 2020-11-09:
> There isn't a dev version in motionEye, only motionEyeOS. Yo umay need to start from scratch on the motion/motionEye install, this time use the Ubuntiu 20.04 and newer instructions. Make sure you use your sudo -i or sudo su before attempting any installation of the software.
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?
ar weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

OOps sorry my mistake I'm probably in the wrong forus its MotioneyeOS that I'm struggling with

starbasessd, 2020-11-09:
You can close this one and open a new one in
https://github.com/ccrisan/motioneyeos/issues
Although I try to help in both, you get different people in the 2 different
setups. I try to reproduce issues, wheter it's in motionEyeOS, or the
motion/motionEye environments to best help.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

icurnow, 2020-11-10:
Thanks I did a fresh install and its now working again. have refrained from updating to dev as this may have caused the issue.

closed 2021-03-05 by starbasessd