motioneye-project/motioneye #1945: Why has mp4 file in the /data/output/Camera1/2020-11-03

jinyistudio, 2020-11-03:
Hi, 
Is mp4 generated by the system itself or after parameter setting?

starbasessd, 2020-11-03:
Depends. If you have movies enabled it generates one of several different
formats;
mkv
mp4
mov
flv
swf
avi
The default is to generate h.264/omx in RPis (not sure of other models) but
can be changed

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-11-03:
Did this answer your question?

closed 2020-11-03 by jinyistudio