motioneye-project/motioneye #1946: no longer cameras in remote browser; but still in local Pi browser

mvh0518, 2020-11-03:
I've had motionEye working for sometime, but about 6 months ago the cameras stopped
working (I have 2 USB cameras and 2 IP cameras)  Yes, it's a lot, but the USB cameras nearly always worked and the IP cameras usually worked. A reboot of this or that always fixed things.
Now instead of pictures I only get the spinning circles.
I've finally gotten around to investigate and I've found that everything works fine
in the browser *running on the local pi* machine (i.e. 127.0.0.1)
Even when the browser is on my windows machine on the local ethernet, I only get 
spinning cameras.  I do get all the MotionEye config options, so I'm interfacing with the motioneye web server.
Did something change in new browsers that impact the camera display in 0.33.4?
What is the next step to investigate?

I'm running the Jessie Raspian build and MotionEye 0.33.4.
FWIW: I've tried to upgrade MotionEye, but it fails with 'Pillow {VERSION} does not support Python'

Thanks
Mark

starbasessd, 2020-11-03:
You really need to upgrade across the board.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-11-03:
motion is up to at least 0.42.1,
motionEye is up to at least 4.2.2
and We're up to Buster for the OS, with major kernel and other OS features
updates.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-11-03:
Closed due to out of date install.

closed 2020-11-03 by starbasessd