motioneye-project/motioneye #1947: Check to see if the disk is recognised:

swaziguy, 2020-11-05:
Check to see if the disk is recognised:

**_sudo dmesg -T_**
(possibly appears as /dev/sda)

**_lsusb_**

If the disk is indeed mounted, look in **_/media/pi/..._**

Here's an example from my own Pi, with a Samsung external HDD:

**_[Tue Oct 20 11:09:48 2020] sd 0:0:0:0: [sda] 312581808 512-byte logical blocks: (160 GB/149 GiB)
[Tue Oct 20 11:09:48 2020] sd 0:0:0:0: [sda] Write Protect is off
[Tue Oct 20 11:09:48 2020] sd 0:0:0:0: [sda] Mode Sense: 28 00 00 00
[Tue Oct 20 11:09:48 2020] sd 0:0:0:0: [sda] No Caching mode page found
[Tue Oct 20 11:09:48 2020] sd 0:0:0:0: [sda] Assuming drive cache: write through
[Tue Oct 20 11:09:48 2020] sd 0:0:0:0: Attached scsi generic sg0 type 0
[Tue Oct 20 11:09:48 2020]  sda: sda1
[Tue Oct 20 11:09:48 2020] sd 0:0:0:0: [sda] Attached SCSI disk_**

_Originally posted by @Johnny-quattro in https://github.com/ccrisan/motioneye/issues/1898#issuecomment-712858201_

Whe I ssh into the pi running motioneye and run the lsusb command I get nothing in return.  I am confused but then again I probably do not understand Linux enough and aaam doing something totally wrong.  Any assistance would be helpful.  All I want to do it record motioneyes video to a locally mounted hard drive but the drive is not showing up in motioneyes.  Does not seem to be mounting.

Johnny-quattro, 2020-11-05:
The key thing here is that the disk appears to not be recognised. If the HDD drive connected via USB is self-powered from the USB cable plugged into your Pi, your Raspberry Pi power adapter may not be able to supply enough current to power it correctly.

Output from 'sudo dmesg -T':

**_[Mon Oct 26 19:25:39 2020] Under-voltage detected! (0x00050005)_**                   
[Mon Oct 26 19:25:41 2020] usb 1-1.3: new high-speed USB device number 6 using dwc_otg
[Mon Oct 26 19:25:41 2020] usb 1-1.3: New USB device found, idVendor=04e8, idProduct=1f01, bcdDevice= 0.00
[Mon Oct 26 19:25:41 2020] usb 1-1.3: New USB device strings: Mfr=1, Product=11, SerialNumber=3
[Mon Oct 26 19:25:41 2020] usb 1-1.3: Product: Samsung S2 Portable
[Mon Oct 26 19:25:41 2020] usb 1-1.3: Manufacturer: JMicron
[Mon Oct 26 19:25:41 2020] usb 1-1.3: SerialNumber: 0000002CE09310500802
[Mon Oct 26 19:25:41 2020] usb-storage 1-1.3:1.0: USB Mass Storage device detected
[Mon Oct 26 19:25:41 2020] scsi host0: usb-storage 1-1.3:1.0
[Mon Oct 26 19:25:41 2020] usbcore: registered new interface driver uas
[Mon Oct 26 19:25:42 2020] scsi 0:0:0:0: Direct-Access     Samsung  S2 Portable           PQ: 0 ANSI: 2 CCS
[Mon Oct 26 19:25:42 2020] sd 0:0:0:0: [sda] 312581808 512-byte logical blocks: (160 GB/149 GiB)
[Mon Oct 26 19:25:42 2020] sd 0:0:0:0: [sda] Write Protect is off
[Mon Oct 26 19:25:42 2020] sd 0:0:0:0: [sda] Mode Sense: 28 00 00 00
[Mon Oct 26 19:25:42 2020] sd 0:0:0:0: [sda] No Caching mode page found
[Mon Oct 26 19:25:42 2020] sd 0:0:0:0: [sda] Assuming drive cache: write through
[Mon Oct 26 19:25:42 2020] sd 0:0:0:0: Attached scsi generic sg0 type 0
[Mon Oct 26 19:25:42 2020]  sda: sda1
[Mon Oct 26 19:25:42 2020] sd 0:0:0:0: [sda] Attached SCSI disk
**_[Mon Oct 26 19:25:43 2020] Voltage normalised (0x00000000)_**

In my case, the disk is successfully mounted. I think I've got one of the older official Raspberry Pi power adapters, which I think may be 1.5A instead of 2A, and I think that may be the reason for my slight dip in voltage.

Output from 'lsusb':

Bus 001 Device 005: ID 1f4d:a803 G-Tek Electronics Group 
**_Bus 001 Device 006: ID 04e8:1f01 Samsung Electronics Co., Ltd_**                              
Bus 001 Device 004: ID 046d:c534 Logitech, Inc. Unifying Receiver
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Output from mount (mounted disks):

mount | grep media
/dev/sda1 on /media/pi/SAMSUNG-160GB type ext4 (rw,nosuid,nodev,relatime,uhelper=udisks2)

I would also check the syslog for error messages:

less /var/log/syslog (use spacebar to display subsequent pages, use q to quit)

starbasessd, 2020-11-05:
Depending on which OS you are running, you may try 
lsusb -t [enter]
If the drive shows with dmesg command, but does not show anything else, it may not be partitioned or formatted.
Most Linux flavors will attempt to auto-mount if it's plugged in when you power up.
If you are using RaspberryPiOS, you can run 
parted /dev/sda [enter]
print [enter]
And it'll give you  info on the partitions if any.

swaziguy, 2020-11-08:
This is what I am getting for an ssh prompt:

[root@Acacia_Farm_CCTV ~]# sudo demes =T
-sh: sudo: command not found
[root@Acacia_Farm_CCTV ~]# lsusb
[root@Acacia_Farm_CCTV ~]# lsusb -t
[root@Acacia_Farm_CCTV ~]# parted /dev/sda
-sh: parted: command not found
[root@Acacia_Farm_CCTV ~]# print
-sh: print: command not found
[root@Acacia_Farm_CCTV ~]#

Johnny-quattro, 2020-11-08:
If you are already root user, then you don't need to use sudo.
demes is not a command and options supplied use the -, not =
dmesg -T or dmesg should work
Since lsusb is not showing anything, it looks as though the Pi is not recognising the USB HDD. Try connecting to another device/PC to see if it actually works.

Full path to run the above commands (if your $PATH is not correct):

/usr/bin/lsusb
/bin/dmesg
/sbin/parted

swaziguy, 2020-11-08:
OK not being a linux expert some of this is very confusing to me but from the little I know I think I am going to the proper path/

OK this is really long but this is what I am getting with each of those commands.  And I did plug in another USB flash drive to the Pi before I did this:

[root@Acacia_Farm_CCTV bin]# /usr/bin/lsusb
[root@Acacia_Farm_CCTV bin]# /bin/dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.83-v7l (root@4bd0e8267adb) (gcc version 8.2.1 20180802 (GNU Toolchain for the A-profile Architecture 8.2-2018.11 (arm-rel-8.26))) #1 SMP Sat Jun 6 14:32:26 UTC 2020
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 64 MiB at 0x0000000034000000
[    0.000000] On node 0 totalpages: 2048000
[    0.000000]   DMA zone: 1728 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 1851392 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0xc0/0x4e0 with crng_init=0
[    0.000000] percpu: Embedded 16 pages/cpu s36748 r8192 d20596 u65536
[    0.000000] pcpu-alloc: s36748 r8192 d20596 u65536 alloc=16*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2046272
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 cma=64M bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:B4:61:24 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.fiq_fix_enable=1 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait ro rootflags=noload panic=10 quiet loglevel=1
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 8037356K/8192000K available (8192K kernel code, 689K rwdata, 2408K rodata, 2048K init, 848K bss, 89108K reserved, 65536K cma-reserved, 7340032K highmem)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xf0800000 - 0xff800000   ( 240 MB)
                   lowmem  : 0xc0000000 - 0xf0000000   ( 768 MB)
                   pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
                   modules : 0xbf000000 - 0xbfe00000   (  14 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (10208 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (2048 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 690 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 849 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 28596 entries in 84 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000005] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000017] Switching to timer-based delay loop, resolution 18ns
[    0.000236] Console: colour dummy device 80x30
[    0.000260] console [tty1] enabled
[    0.000302] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.000324] pid_max: default: 32768 minimum: 301
[    0.000588] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000606] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001365] CPU: Testing write buffer coherency: ok
[    0.001766] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002412] Setting up static identity map for 0x200000 - 0x20003c
[    0.002569] rcu: Hierarchical SRCU implementation.
[    0.003433] smp: Bringing up secondary CPUs ...
[    0.004290] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.005280] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.006231] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.006366] smp: Brought up 1 node, 4 CPUs
[    0.006378] SMP: Total of 4 processors activated (432.00 BogoMIPS).
[    0.006388] CPU: All CPU(s) started in HYP mode.
[    0.006397] CPU: Virtualization extensions available.
[    0.007151] devtmpfs: initialized
[    0.017619] VFP support v0.3: implementor 41 architecture 3 part 40 variant 8 rev 0
[    0.017841] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.017861] futex hash table entries: 1024 (order: 4, 65536 bytes)
[... 426 lines cut ...]
[84184.237776] FS-Cache: O-cookie d=85b315b1 n=f2fe0275
[84184.237780] FS-Cache: O-key=[9] '4343545653746f7265'
[84184.237819] FS-Cache: N-cookie c=7b40038c [p=287340b5 fl=2 nc=0 na=1]
[84184.237824] FS-Cache: N-cookie d=85b315b1 n=53c64482
[84184.237827] FS-Cache: N-key=[9] '4343545653746f7265'
[84229.724209] FS-Cache: Duplicate cookie detected
[... cut ...]
[84229.866888] FS-Cache: Duplicate cookie detected
[84229.866903] FS-Cache: O-cookie c=230afbd7 [p=287340b5 fl=222 nc=0 na=1]
[84229.866907] FS-Cache: O-cookie d=85b315b1 n=f2fe0275
[84229.866912] FS-Cache: O-key=[9] '4343545653746f7265'
[84229.866950] FS-Cache: N-cookie c=7b40038c [p=287340b5 fl=2 nc=0 na=1]
[84229.866954] FS-Cache: N-cookie d=85b315b1 n=b747ae56
[84229.866958] FS-Cache: N-key=[9] '4343545653746f7265'
[root@Acacia_Farm_CCTV bin]# /sbin/parted
-sh: /sbin/parted: No such file or directory
[root@Acacia_Farm_CCTV bin]#

starbasessd, 2021-03-05:
If you are still having issues, please open a new issue.

closed 2021-03-05 by starbasessd