motioneye-project/motioneye #1951: RPi 4 4G Web Gui does not load

soflyj, 2020-11-16:
Hi

There are several instances where I would make some change to the software and after reboot the GUI wouldn't load even though the IP address is registered on my network. I am connected via network port/ ethernet.

- Removing a camera
- Changing the resolution

Please can you assist.

I have downloaded and attached all logs
[motioneye (1).log](https://github.com/ccrisan/motioneye/files/5548799/motioneye.1.log)
[dmesg.log](https://github.com/ccrisan/motioneye/files/5548796/dmesg.log)
[boot.log](https://github.com/ccrisan/motioneye/files/5548797/boot.log)
[messages.log](https://github.com/ccrisan/motioneye/files/5548798/messages.log)

starbasessd, 2020-11-16:
Looks like you are on  192.168.1.101, is that the IP you are using? I
noticed when first set up it was 192.168.1.12
You are using motionEyeOS 20200606. You may want to upgrade to dev20201026.
I had a lot of issues with 20200606 with RPi4-4GB & RPi4-8GB.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

soflyj, 2020-11-16:
Thanks @starbasessd. I'm going to download and install. I'll keep you updated.

Should I download the os file ([](https://github.com/ccrisan/motioneyeos/releases)) or install via the terminal ([](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian))?

starbasessd, 2020-11-16:
Cleaner to install the file.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

soflyj, 2020-11-16:
So I installed the latest now the setting on the side get randomly disabled (greyed out) and when I can change settings on a camera I don't see the save changes.

Can't even get you the logs
[image]

starbasessd, 2020-11-16:
Before you click on the settings in the upper left, click on the icon for
user, log into motionEye as 'admin' and either no password, or whatever
password you set.
If that doesn't work, I'm betting either it's a corrupted image or bad
SDCard...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

soflyj, 2020-11-16:
> Before you click on the settings in the upper left, click on the icon for user, log into motionEye as 'admin' and either no password, or whatever password you set. If that doesn't work, I'm betting either it's a corrupted image or bad SDCard...
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

Sorr

closed 2020-11-16 by soflyj