motioneye-project/motioneye #1952: Repeated camera image on two cameras???

gardiol, 2020-11-22:
Hi!
i have a setup with some 10 cameras. 
All network cameras. Some Axis, some Dahua. 
The Dahuas are over ONVIF, the AXIS over http static image grab.

Today i added a new dahua camera (configured in motioneye as url: rtsp://192.168.1.243:554/cam/realmonitor?channel=1&subtype=0&unicast=true&proto=Onvif). 

Well, it has the exact same image as the previous camera in the list! (url: rtsp://192.168.1.244:554/cam/realmonitor?channel=1&subtype=0&unicast=true&proto=Onvif)! I mean, i can see the two camer boxes in the web page, with correct overlys (camera title is correct) but they both display the exact same image And they are both live too!

If i "swap" them in the motion.conf by hand, well, i also swap the displayed camera. It seems that the first one in motion.conf is used for both cameras in the web page?

The two cameras have different ID (11 and 12) and different name...

motionEye Version | 0.42.1
Motion Version | 4.3.1

OS is Linux.

Can somebody help be out on this? 
In the logs (debug level) nothing strange is displayed... 
This is not the first time it happens, but usually playing a bit around in the conf files, renaming them ans such helps. 

This time i would like to fix it o understand why for good!!!

starbasessd, 2020-11-22:
Which Linux OS?
Which installation instructions?
Are the Camera numbers in sequence?
If you re-organize the camera names and numbers to be insequence, does that
fix it? (When I ran 8 cameras and had that issue, I re-sequenced mine and
it fixed it.)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

gardiol, 2020-11-22:
Gentoo Linux.
I have installed with pip2 install motioneye

The cameras are not in sequence.

Motion.conf:
`
camera camera-3.conf
camera camera-5.conf
camera camera-4.conf
camera camera-6.conf
camera camera-10.conf
camera camera-11.conf
camera camera-12.conf
camera camera-13.conf
`

I can try to reorganize them, i will try and report later.

starbasessd, 2020-11-22:
Let me know.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-11-22:
32 or 64 bit?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

gardiol, 2020-11-22:
No, it did not work.
I have copied all camera files, leaving only the 8 active ones. Renamed them from camera-1.conf to camera-8.conf, fixed the @id inside to match the file name, added them to motion.conf. 

Nothing changes, still the same issue.

This is the first camera file (camera-5.conf):

```
# @webcam_resolution 100
# @upload_subfolders on
# @upload_server 
# @enabled on
# @network_server 
# @motion_detection off
# @upload_port 
# @upload_location 
# @preserve_movies 30
# @network_username 
# @upload_movie on
# @id 5
# @webcam_server_resize off
# @upload_password 
# @manual_record off
# @upload_method post
# @upload_picture on
# @working_schedule_type outside
# @network_password 
# @upload_service ftp
# @preserve_pictures 30
# @storage_device custom-path
# @manual_snapshots on
# @network_share_name 
# @upload_enabled off
# @network_smb_ver 1.0
# @working_schedule 
# @clean_cloud_enabled off
# @upload_username 

threshold_maximum 0
netcam_userpass xxx:xxx
stream_quality 85
threshold 7996
noise_level 31
smart_mask_speed 0
pre_capture 1
movie_codec mp4:h264_v4l2m2m
noise_tune on
stream_maxrate 5
netcam_url rtsp://192.168.1.243:554/cam/realmonitor?channel=1&subtype=0&unicast=true&proto=Onvif
stream_localhost on
text_changes off
movie_filename %Y-%m-%d/%H-%M-%S
movie_max_time 0
lightswitch_percent 0
movie_passthrough on
event_gap 30
auto_brightness off
stream_port 8091
rotate 0
stream_auth_method 0
threshold_tune off
framerate 2
emulate_motion off
movie_output on
picture_quality 85
snapshot_filename %Y-%m-%d/%H-%M-%S
despeckle_filter
snapshot_interval 0
minimum_motion_frames 20
stream_motion off
target_dir /webcam/motioneye/Cucina
movie_output_motion off
post_capture 1
stream_authentication user:
on_picture_save /home/motioneye/.local/lib64/python2.7/site-packages/motioneye/scripts/relayevent.sh "/home/motioneye/conf/motioneye.conf" picture_save %t %f
on_movie_end /home/motioneye/.local/lib64/python2.7/site-packages/motioneye/scripts/relayevent.sh "/home/motioneye/conf/motioneye.conf" movie_end %t %f
text_left Cucina
picture_output_motion off
picture_filename %Y-%m-%d/%H-%M-%S
text_scale 1
locate_motion_style redbox
locate_motion_mode off
netcam_keepalive off
movie_quality 75
picture_output on
on_event_end /home/motioneye/.local/lib64/python2.7/site-packages/motioneye/scripts/relayevent.sh "/home/motioneye/conf/motioneye.conf" stop %t
text_right %Y-%m-%d\n%T
on_event_start /home/motioneye/.local/lib64/python2.7/site-packages/motioneye/scripts/relayevent.sh "/home/motioneye/conf/motioneye.conf" start %t
camera_name Cucina
netcam_tolerant_check on
height 960
mask_file
width 1280
```

This is the second camera file (camera-8.conf):

```
# @webcam_resolution 100
# @upload_subfolders on
# @upload_server 
# @enabled on
# @network_server 
# @motion_detection off
# @upload_port 
# @upload_location 
# @preserve_movies 30
# @network_username 
# @upload_movie on
# @id 8
# @webcam_server_resize off
# @upload_password 
# @manual_record off
# @upload_method post
# @upload_picture on
# @working_schedule_type outside
# @network_password 
# @upload_service ftp
# @preserve_pictures 30
# @storage_device custom-path
# @manual_snapshots on
# @network_share_name 
# @upload_enabled off
# @network_smb_ver 1.0
# @working_schedule 
# @clean_cloud_enabled off
# @upload_username 

threshold_maximum 0
netcam_userpass xxx:xxx
stream_quality 85
threshold 7996
noise_level 31
smart_mask_speed 0
pre_capture 1
movie_codec mp4:h264_v4l2m2m
noise_tune on
stream_maxrate 5
netcam_url rtsp://192.168.1.244:554/cam/realmonitor?channel=1&subtype=0&unicast=true&proto=Onvif
stream_localhost on
text_changes off
movie_filename %Y-%m-%d/%H-%M-%S
movie_max_time 0
lightswitch_percent 0
movie_passthrough on
event_gap 30
auto_brightness off
stream_port 8091
rotate 0
stream_auth_method 0
threshold_tune off
framerate 2
emulate_motion off
movie_output on
picture_quality 85
snapshot_filename %Y-%m-%d/%H-%M-%S
despeckle_filter
snapshot_interval 0
minimum_motion_frames 20
stream_motion off
target_dir /webcam/motioneye/StradaSud
movie_output_motion off
post_capture 1
stream_authentication user:
on_picture_save /home/motioneye/.local/lib64/python2.7/site-packages/motioneye/scripts/relayevent.sh "/home/motioneye/conf/motioneye.conf" picture_save %t %f
on_movie_end /home/motioneye/.local/lib64/python2.7/site-packages/motioneye/scripts/relayevent.sh "/home/motioneye/conf/motioneye.conf" movie_end %t %f
text_left StradaSud
picture_output_motion off
picture_filename %Y-%m-%d/%H-%M-%S
text_scale 1
locate_motion_style redbox
locate_motion_mode off
netcam_keepalive off
movie_quality 75
picture_output on
on_event_end /home/motioneye/.local/lib64/python2.7/site-packages/motioneye/scripts/relayevent.sh "/home/motioneye/conf/motioneye.conf" stop %t
text_right %Y-%m-%d\n%T
on_event_start /home/motioneye/.local/lib64/python2.7/site-packages/motioneye/scripts/relayevent.sh "/home/motioneye/conf/motioneye.conf" start %t
camera_name StradaSud
netcam_tolerant_check on
height 960
mask_file
width 1280
```

With this motion.conf i see only cmera5 repeated twice:

```
# @normal_password 
# @enabled on
# @admin_username admin
# @show_advanced on
# @admin_password 
# @normal_username user

setup_mode off
webcontrol_parms 2
webcontrol_localhost off
webcontrol_port 7999

webcontrol_interface 1

camera camera-1.conf
camera camera-2.conf
camera camera-3.conf
camera camera-4.conf
camera camera-5.conf
camera camera-6.conf
camera camera-7.conf
camera camera-8.conf
```

While with this motion.conf, i see only camera8 twice:

```
# @normal_password 
# @enabled on
# @admin_username admin
# @show_advanced on
# @admin_password 
# @normal_username user

setup_mode off
webcontrol_parms 2
webcontrol_localhost off
webcontrol_port 7999

webcontrol_interface 1

camera camera-1.conf
camera camera-2.conf
camera camera-8.conf
camera camera-3.conf
camera camera-4.conf
camera camera-5.conf
camera camera-6.conf
camera camera-7.conf
```

Basically, the first one of the two...

starbasessd, 2020-11-22:
Which one is missing of the other 7?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

gardiol, 2020-11-22:
Camera 5 and  camera 8 share the same image. The first one in motion.conf is displayed twice, the second one is missing. The overlay is correct (each camera in the web gui has the correct camera name... but the second one will have the same image of the first).

starbasessd, 2020-11-22:
What is your gentoo setup? 32 or 64 bit?
What was your base image before installing motion/motionEye?
Which set of instructions did you use to install motion/motionEye?
I am building a VM to reproduce your issue(s)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

gardiol, 2020-11-22:
My setup is a Gentoo box, 64bit, AMD64 is the architecture. I use motioneye with NGINX as reverse proxy.
My base image? I have a standard Gentoo manual installation (profile 17.1, reasonably up-to-date environment).

I have created a "motioneye" user, all files are under /home/motioneye. I usually do:
`sudo -u motioneye bash`

And then:
`pip2 install motioneye`

Also i do this from time to time to ensure motioneye is updated.

Motion, on the other hand, i have downloaded the 4.3.1 tar bar from:
[https://github.com/Motion-Project/motion/releases](https://github.com/Motion-Project/motion/releases)
and manually compiled / installed it following the doc/INSTALL instruction. I have changed the default prefix:
`./configure --prefix=/home/motioneye/.local`
I have compiled and installed as motioneye user, not root.

I have the following motioneye.conf file:

```
# path to the configuration directory (must be writable by motionEye)
conf_path /home/motioneye/conf

# path to the directory where pid files go (must be writable by motionEye)
run_path /var/run/motioneye

# path to the directory where log files go (must be writable by motionEye)
log_path /var/log/motioneye

# default output path for media files (must be writable by motionEye)
media_path /webcam/motioneye/

# the log level (use quiet, error, warning, info or debug)
log_level error

# the IP address to listen on
# (0.0.0.0 for all interfaces, 127.0.0.1 for localhost)
listen 0.0.0.0

# the TCP port to listen on
port 8765

# path to the motion binary to use (automatically detected if commented)
motion_binary /home/motioneye/.local/bin/motion

# whether motion HTTP control interface listens on
# localhost or on all interfaces
motion_control_localhost true

# the TCP port that motion HTTP control interface listens on
motion_control_port 7999

# interval in seconds at which motionEye checks if motion is running
motion_check_interval 10

# whether to restart the motion daemon when an error occurs while communicating with it
motion_restart_on_errors false

# interval in seconds at which motionEye checks the SMB mounts
mount_check_interval 300

# interval in seconds at which the janitor is called
# to remove old pictures and movies
cleanup_interval 36000

# timeout in seconds to wait for response from a remote motionEye server
remote_request_timeout 10

# timeout in seconds to wait for mjpg data from the motion daemon
mjpg_client_timeout 10

# timeout in seconds after which an idle mjpg client is removed
# (set to 0 to disable)
mjpg_client_idle_timeout 10

# enable SMB shares (requires motionEye to run as root) 
smb_shares false

# the directory where the SMB mount points will be created
smb_mount_root /media

# path to the wpa_supplicant.conf file
# (enable this to configure wifi settings from the UI)
#wpa_supplicant_conf /etc/wpa_supplicant.conf

# path to the localtime file
# (enable this to configure the system time zone from the UI)
#local_time_file /etc/localtime

# enables shutdown and rebooting after changing system settings
# (such as wifi settings or time zone)
enable_reboot false

# timeout in seconds to use when talking to the SMTP server
smtp_timeout 60

# timeout in seconds to wait for media files list
list_media_timeout 120

# timeout in seconds to wait for media files list, when sending emails
list_media_timeout_email 10

# timeout in seconds to wait for zip file creation
zip_timeout 500

# timeout in seconds to wait for timelapse creation
timelapse_timeout 500

# enable adding and removing cameras from UI
add_remove_cameras true

# enables HTTP basic authentication scheme (in addition to, not instead of the signature mechanism)
http_basic_auth false

# overrides the hostname (useful if motionEye runs behind a reverse proxy)
# server_name motionEye
```

I can provide also the other 6 missing camera-X.conf files if needed.

I also have the following start script:
`start-stop-daemon -b -m -p /var/run/motioneye/motioneye.pid -n motioneye -u motioneye /home/motioneye/.local/bin/meyectl -- startserver -c /home/motioneye/conf/motioneye.conf`

to run motioneye.

gardiol, 2020-11-22:
This is my NGINX config related to motioneye:
```
                location /me/ {
                        proxy_pass http://127.0.0.1:8765/;
                        proxy_read_timeout 120s;
                        access_log off;
                }

```

starbasessd, 2020-11-22:
Did you use the make config file with the motioneye mods used by motioneye?
There are a number of mods made for motion to work with motioneye.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

gardiol, 2020-11-22:
I am checking the requests / responses on firefox developer console, and they seems correct.

I can see the request:
` http://192.168.1.1:8765/picture/8/current/?_=1606076593234&width=0.9&_username=admin&_signature=fb61c4435c522da3e63d4be791ae8a0d512139bf `
and the request:
` http://192.168.1.1:8765/picture/5/current/?_=1606076593234&width=0.9&_username=admin&_signature=e3aa3b31a352ea970cb290a9e4687efc2f145936`

The response is the exact identical image.

gardiol, 2020-11-22:
> Did you use the make config file with the motioneye mods used by motioneye? There are a number of mods made for motion to work with motioneye.
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

No i did not...
Where do i find it?

starbasessd, 2020-11-22:
@jasaw or @ccrisan do you have the modified compile configuration file for motion lying around here somewhere? User needs to manually compile motion for Gentoo Linux...

starbasessd, 2020-11-22:
I forgot how much I hate compiling Gentoo... Unless I need a hardened system and need to view the source code for each system/application, I don't see the advantage to a completely manually compiled system. There are many SELinux systems out there. If you are using the base 17.1, you don't even get that advantage.
I am just now completing the emerge --ask --verbose --update --deep --newuse @world command...
:(

zagrim, 2020-11-23:
@gardiol The cams seem to have the same `stream_port` (8091), which messes up at least the display of the images in the GUI, don't know if that causes any issues in the saved files, though. Please try to manually change the streaming port value of either of the cams (under Video Streaming > Streaming Port in the ME GUI).

gardiol, 2020-11-23:
Yes! It was the stream_port! Ensuring each camera has a different one fixed the issue! 
Thank you...

I can also confirm that saved files are not impacted by this problem. It all makes sense now.

gardiol, 2020-11-23:
`I forgot how much I hate compiling Gentoo... `

To be honest, i use gentoo only since it's creation, and i am quite pleased. Yes maybe it's a bit a pain on daily driver work laptops but 100% would never install&manage a non-Gentoo server. It's not a matter of features, but of ease of management,  rock solid stability, no weird behaviours whatsoever, and stable unchanged software unless there is a really good reason not to.  In addition to all that, it's basically bleeding-edge and very much updated to the latest releases. 

With today's CPUs, the compile times have dwindled down to nothing (compared to 40+ hours to compile Firefox... or a week to compile OpenOffice...).

@starbasessd , you can take down the VM and stop compiling :)

closed 2020-11-23 by gardiol