motioneye-project/motioneye #1953: Motion detection not working at night with Pi Noir v2 camera

jcohenho, 2020-11-23:
I'm using a Pi Noir v2 camera paired with my Pi 3. I am using an IR illuminator along with the camera and I can see very clearly via the web interface, but no matter what settings I try, the camera will not capture any motion at night. Are there some settings or other configurations I need to enable? 

Here is a screenshot of my current settings and picture quality: 

[image]

[image]

starbasessd, 2020-11-23:
I have my cameras with despeckle ON and frame change threshold at 0.1% and
it's a bit too sensitive. I use 2,10,10, 2 for gap, before, after, and
capture.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jcohenho, 2020-11-23:
@starbasessd I updated my config to use these settings and still no luck. Did you have to add anything to 'Extra Motion Options' or adjust the camera brightness?

starbasessd, 2020-11-23:
Nope. Which setup instructions did you use and what base OS are you using?
Are you able to capture anything in a well-light (daylight-type)
environment?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jcohenho, 2020-11-23:
@starbasessd I cannot capture anything day or night with the Noir, but manually taking snapshots does work. I was previously motion capturing everything just fine with my Rasperry Pi camera module v2. Swapping it out with the Noir and now capturing doesn't work. I am running on motioneyeOS, I did a basic installation - formatted an SD card and wrote the motioneyeOS image to it.

starbasessd, 2020-11-23:
Which version? 20190911, 20200203, 20200606, or dev20201026?
You may want to close this issue, and open one at
https://github.com/ccrisan/motioneyeos/issues

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jcohenho, 2020-11-23:
@starbasessd I'm using 20200606. Okay, will do thanks for all the info.

starbasessd, 2020-11-23:
BTW, try 20190911 or dev20201026...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2020-11-24 by starbasessd

jcohenho, 2021-01-02:
@starbasessd I re-installed the same OS version and it works now...maybe it was buggy swapping the cameras out without a re-install. Thanks again for your help.