motioneye-project/motioneye #1956: Motion notification via email - no picture attached

mireksochor, 2020-11-24:
Hi and thank you for such a great piece of SW, really appreciate the hard work you put into MotionEye.
I have fresh install in Rpi 3B, all configured, working well except no pictures attached to email.
My settings is:

motionEye Version | 0.42.1
Motion Version | 4.2.2+gitUNKNOWN
OS Version | motionEyeOS 20200606

starbasessd, 2020-11-24:
Please include logs: dmesg, messages.log, motioneye.log.
Do you have a lot of pictures being stored on the Pi in question?
Also consider using 20190911, 20200203, or dev20201026.
The motionEyeOS support pages are over at
Home · ccrisan/motioneyeos Wiki
<https://github.com/ccrisan/motioneyeos/wiki>

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mireksochor, 2020-11-25:
Hi! 
there is less than 20 pictures stored.
Logs bellow:
**motioneye.log**
2020-11-24 12:28:42: [sendmail]     INFO: sending email
2020-11-24 12:28:44: [sendmail]     INFO: email sent
2020-11-24 13:35:14: [sendmail]     INFO: sending email
2020-11-24 13:35:15: [sendmail]     INFO: email sent
2020-11-24 13:35:56: [sendmail]     INFO: sending email
2020-11-24 13:35:57: [sendmail]     INFO: email sent
2020-11-24 17:10:30: [motioneye]  WARNING: 403 GET /login/?_=1606234230239&_username=_&_signature=3b4750a5429cacaaefcad27bdfd4be2558fc029f (10.0.1.171) 5.02ms
2020-11-24 17:25:06: [motioneye]  WARNING: mjpg client connection for camera 2 on port 8082 is closed

**dmesg.log**
---- booting motionEyeOS 20200606 ----
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.83-v7 (root@a49a11003f4f) (gcc version 8.2.1 20180802 (GNU Toolchain for the A-profile Architecture 8.2-2018.11 (arm-rel-8.26))) #1 SMP Sat Jun 6 14:14:33 UTC 2020
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x2f800000
[    0.000000] On node 0 totalpages: 196608
[    0.000000]   Normal zone: 1728 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 196608 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0xb0/0x4b4 with crng_init=0
[    0.000000] percpu: Embedded 16 pages/cpu s36684 r8192 d20660 u65536
[    0.000000] pcpu-alloc: s36684 r8192 d20660 u65536 alloc=16*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 194880
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.fiq_fix_enable=1 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait ro rootflags=noload panic=10 quiet loglevel=1
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 756380K/786432K available (8192K kernel code, 655K rwdata, 2228K rodata, 1024K init, 820K bss, 21860K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xb0800000 - 0xff800000   (1264 MB)
                   lowmem  : 0x80000000 - 0xb0000000   ( 768 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (9184 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 656 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 821 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 26874 entries in 79 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000018] Switching to timer-based delay loop, resolution 52ns
[    0.000280] Console: colour dummy device 80x30
[    0.000299] console [tty1] enabled
[    0.000338] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000355] pid_max: default: 32768 minimum: 301
[    0.000676] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000691] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001627] CPU: Testing write buffer coherency: ok
[    0.002086] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002732] Setting up static identity map for 0x100000 - 0x10003c
[    0.002883] rcu: Hierarchical SRCU implementation.
[    0.003671] smp: Bringing up secondary CPUs ...
[    0.004505] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.005446] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.006307] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.006432] smp: Brought up 1 node, 4 CPUs
[    0.006443] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[... 224 lines cut ...]
Nov 24 17:17:36 meye-e8e4ef18 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
Nov 24 17:17:36 meye-e8e4ef18 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2+gitUNKNOWN Started
Nov 24 17:17:36 meye-e8e4ef18 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Nov 24 17:17:36 meye-e8e4ef18 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Nov 24 17:18:16 meye-e8e4ef18 user.notice throttlewatch: currently: under-voltage throttled , temperature: 40 C
[... cut ...]
Nov 24 17:25:04 meye-e8e4ef18 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2+gitUNKNOWN Started
Nov 24 17:25:04 meye-e8e4ef18 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Nov 24 17:25:04 meye-e8e4ef18 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Nov 24 17:25:17 meye-e8e4ef18 user.notice throttlewatch: currently: under-voltage throttled , temperature: 41 C
Nov 24 17:26:17 meye-e8e4ef18 user.notice throttlewatch: currently: under-voltage throttled , temperature: 39 C

starbasessd, 2020-11-25:
In messages.log first thing I notice is you have a bad/underpowered power
supply. Throttling causes all kinds of problems. The board is trying to
protect itself.
If you can ssh in or use the console. run the command:
logrotate -f /etc/logrotate.conf [enter]
Wait 24 hours, then download and attach the complete log files to the
\https://github.com/ccrisan/motioneye/issues/1956
incident ticket. You can download the logs from the Settings, Expert
Settings, section of the WebGUI.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mireksochor, 2020-11-25:
Thank you for pointing that out @starbasessd. The underpower is quite unprobable as I tried 3 microUSB cables and 2 power bricks (18W apple charger and 12W power charger) and Raspi never take more than 700mA on 5.2V when I tested. It has small fan installed, but no cooler. I will try previous OS release and see if I can make it work.

starbasessd, 2020-11-25:
"Chargers" are never a good thing(TM). They have current limiting circuitry
and charging circuitry built in that the PI can't talk to. (especially
Apple products).
I have a number of official RPi power supplies, and a number of other power
supplies that don't have issues (2.5a or larger) 3 'chargers' (15 W or
better) all have issues with Pis, even PiZero. All work to spec with
devices requiring 'chargers'.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

zagrim, 2020-11-26:
Just a thought about the undervoltage and the fan you mentioned: Even a small fan drains relatively lot of current, and if the fan is powered through the Pi it still might overload the capacity of the Pi. Myself I had stability issues on a Pi2 with a extension board (powered through the Pi2) and a USB Wifi adapter until I plugged the extension board to a separate power supply even though I was already using a USB power supply with the output something like 2.1 - 2.5 A (10.5 - 12.5). Pi2 uses less power itself, but might also be more limited than Pi3 WRT supplying power to other devices, so this will not directly match your specs, but hopefully makes my point clear, though.

zagrim, 2020-11-26:
Regarding the topic issue, there's this: https://github.com/ccrisan/motioneyeos/wiki/FAQ#what-should-i-do-to-have-pictures-attached-to-motion-notification-emails
I have myself also experienced this, most of the notifications never had any picture, and as far as I've understood that's because the email is sent before the snapshot image is produced in some cases. I'm pretty sure there's been a issue explaining the details around that but I can't find it now. Basically that would be due to the snapshot being saved to the disk only after the capture ends (in case of saving the "best" picture) and there being a sort of a race condition between that and the email sending process.

mireksochor, 2020-11-26:
Thanks @zagrim, I appreciate your explanation. I have found the power source that is just 1.5A but RasPi is not complaining. 
You very well described my thoughts, thank you for that, regarding the timing of email being sent and snapping the photo. Is there any way how to set wait time for the photo? Or make and condition that email without attachment wont leave?

zagrim, 2020-11-26:
My explanation was still only guessing based on something someone has written here, at some point (now that was vague, uh :) ). Anyway, since it is an extenal process (Motion) running which is saving the files, in continuously parallel with ME, it's not that easy to tell when is has finished its job. Unless I remember it wrong, when ME is configured with Capture Mode = Motion Triggered (One Picture), what Motion does is to scan through to captured frames to find the one with highest amount of changed pixels and then save that. On a RPi that might take a while, depending on the framerate and video length & quality.

One thing you could try is to increase the "Attached Pictures Time Span" under notification settings to see if that would be a key to success. At least that setting should directly impact how long the email sending waits before searching for files to attach.

If that doesn't help, the only thing I can come up with at the moment is to increase logging verbosity to "debug" (`log_level` in `motioneye.conf`) and then inspect the `motioneye.log` (attach it here) after motion has been detected - and the changing the log level back to "info" to avoid bloating the logs. From the logs it should be, hopefully, possible to see if the email is actually sent before the picture has been saved.

starbasessd, 2020-11-26:
in /etc/motioneye/motioneye.conf there is a parameter:
list_media_timeout_email 10
that you could increase by 10 until you get consistent attachments...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

zagrim, 2020-11-27:
> list_media_timeout_email 10

Yes, that, too, although the logs should indicate with an error if that timeout is reached when searching for media files.

closed 2021-03-05 by starbasessd