motioneye-project/motioneye #1958: Dropbox: Obtain Key leads to 'Invalid client_id: "dropbox_client_id_placeholder"'

kristianrl, 2020-11-30:
I'm trying to set up Dropbox upload of my MotionEye recordings.

I clicked _Obtain Key_, while logged in as admin:
[image]
and it sends me to the URL _https://www.dropbox.com/oauth2/authorize?response_type=code&client_id=dropbox_client_id_placeholder_

> Error (400)
> It seems the app you were using submitted a bad request. If you would like to report this error to the app's developer, include the information below.
> 
> 
> More details for developers
> Invalid client_id: "dropbox_client_id_placeholder".

I have tried while being logged in to Dropbox, as well as in an incognito session while logged out of Dropbox. 

I could imagine that someplace in the code, `dropbox_client_id_placeholder` is hardcoded instead of an actual Dropbox client ID.

starbasessd, 2020-11-30:
What version of motion & motionEye are you running?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

kristianrl, 2020-11-30:
Hi @starbasessd , I'm using the [most recent master-amd64 Docker build](https://hub.docker.com/layers/ccrisan/motioneye/master-amd64/images/sha256-1704cddc44e0bc7e1c4aff9f1f6546127584e8351c04a0faa1d46c7fb791180b?context=explore):

motionEye Version | 0.42
Motion Version | 4.2.2
OS Version | Ubuntu 18.10

starbasessd, 2020-11-30:
I am setting up a Dropbox connection on a dev20201026, hang on...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-11-30:
> I'll then check the Docker version.
>
> On Mon, Nov 30, 2020 at 4:38 PM Kevin Shumaker <kevin.shumaker@gmail.com>

Set up basic DropBox account.
Installed local DropBox app for Win10
Set up /camtest/ folder
Requested Dropbox key using button, installed, working.
Now checking a docker install in a few minutes (gotta build a Ubuntu Server
for the Docker.)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-11-30:
Confirm issue on docker image in Ubuntu 18.04 host (fully updated) (amd64) and RaspberryPiOS (fully updated) (armhf).
[image]
Dropbox works in any 'normal' (read: non-docker) install (Tested Hosts: Ubuntu 18.04, Ubuntu 20.04 & 20.10, RaspberryPiOS 2020-08-20, and motionEyeOS dev20201026.

starbasessd, 2020-11-30:
Now doing a BYOD (Build Your Own Docker) install on a P4-4GB to see what's
current in the dev branch
Previous tests were don with the command(s) found here:
https://github.com/ccrisan/motioneye/wiki/Install-In-Docker
AMD64:

docker run --name="motioneye" -p 8765:8765 --hostname="motioneye" -v
/etc/localtime:/etc/localtime:ro -v /etc/motioneye:/etc/motioneye -v
/var/lib/motioneye:/var/lib/motioneye --restart="always" --detach=true
ccrisan/motioneye:master-amd64

and ARMHF

docker run --name="motioneye" -p 8765:8765 --hostname="motioneye" -v
/etc/localtime:/etc/localtime:ro -v /etc/motioneye:/etc/motioneye -v
/var/lib/motioneye:/var/lib/motioneye --restart="always" --detach=true
ccrisan/motioneye:master-armhf

Now running the BYOD using the command:

docker build --build-arg VCS_REF=$(git rev-parse HEAD) --build-arg
BUILD_DATE=$(date +"%Y-%m-%dT%H:%M:%SZ") -t
ccrisan/motioneye:master-armhf -f extra/Dockerfile.armv7-armhf .

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-11-30:
Even the BYOD (Build Your Own Docker) has the issue. 
Tagged as bug in Docker. @ccrisan  notified.

starbasessd, 2020-11-30:
As a work-around, I can recommend installing motion/motionEye in the host system (use the Ubuntu https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29 instructions for any Debian-type install)

starbasessd, 2020-12-01:
I'll then check the Docker version.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sukanth, 2021-01-04:
I have the same issue.. any idea on when it will be fixed?  @ccrisan @starbasessd

starbasessd, 2021-01-08:
I just tested, and it worked fine for me.
Took me into the sign-on page, I  used my account info, and it gave me a
key.
I copied/pasted the key, clicked apply, then clicked test and it was fine.
Using motion 4.3.1,
motionEye 0.42.1 on
Ubuntu 20.04, all fully updated.
Also tested on RPiZeroW 20200606, same result.
Be sure to clear your browser cache before trying.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sukanth, 2021-01-08:
it worked fine for me on ther motioneyeOS but not in the motioneye UI

starbasessd, 2021-01-08:
Just tested on my production hub (previous test was on a purpose built test
VM)
same steps, same successful result.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sukanth, 2021-01-08:
This is what happens for me when i click on ObtainKey
[image]

starbasessd, 2021-01-08:
I assume you already have a Dropbox account?
Did you clear cache in whichever browser is triggered?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sukanth, 2021-01-08:
this is the version i am running on
[image]

sukanth, 2021-01-08:
> I assume you already have a Dropbox account? Did you clear cache in whichever browser is triggered?
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

Yes i do have dropbox and did clear the cache several times and tried different browsers as well

starbasessd, 2021-01-08:
Upgrade to motion 4.3.1 with the following commands:

get https://github.com/Motion-Project/motion/releases/download/release-4.3.1/pi_buster_motion_4.3.1-1_armhf.deb
<https://github.com/Motion-Project/motion/releases/download/release-4.2.2/pi_buster_motion_4.2.2-1_armhf.deb>
dpkg -i pi_buster_motion_4.3.1-1_armhf.deb

 pip install motioneye --upgrade
 systemctl restart motioneye

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sukanth, 2021-01-08:
> Upgrade to motion 4.3.1 with the following commands: get https://github.com/Motion-Project/motion/releases/download/release-4.3.1/pi_buster_motion_4.3.1-1_armhf.deb <https://github.com/Motion-Project/motion/releases/download/release-4.2.2/pi_buster_motion_4.2.2-1_armhf.deb> dpkg -i pi_buster_motion_4.3.1-1_armhf.deb pip install motioneye --upgrade systemctl restart motioneye On Fri, Jan 8, 2021 at 4:31 PM Sukanth Gunda <notifications@github.com> wrote:
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

do you have a docker container for this version.. i am running it inside docker..

sukanth, 2021-01-31:
@starbasessd  i am running motioneye on raspberry pi using a docker container .. how do i upgrade the motion version inside docker? 
[image]
[image]

starbasessd, 2021-01-31:
Which Pi? 3B, 3B+, 4B-4GB, or 4B-8GB?
Which Docker install? I normally use docker.io
Which install of the motionEye Docker?
https://github.com/ccrisan/motioneye/wiki/Install-In-Docker Image or Build?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-01-31:
Oh, and what base os? RaspberryPiOS, experimental RaspberryPiOS-64, Ubuntu
Server or Desktop? Something else?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sukanth, 2021-02-01:
> Which Pi? 3B, 3B+, 4B-4GB, or 4B-8GB? Which Docker install? I normally use docker.io Which install of the motionEye Docker? https://github.com/ccrisan/motioneye/wiki/Install-In-Docker Image or Build?
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

@starbasessd 
I am using Raspberry Pi-3B Model. 
I am using this image "ccrisan/motioneye:master-armhf" to build the container.

this is the command that i am using to build the container.

docker run --name="motioneye" -p 8999:8999 --hostname="motioneye" -v /etc/localtime:/etc/localtime:ro -v /etc/motioneye:/etc/motioneye -v /var/lib/motioneye:/var/lib/motioneye -v /dev/video0:/dev/video0 -e TZ=America/Chicago --restart="always" --detach=true --device=/dev/video0 ccrisan/motioneye:master-armhf

the base OS that I am using is Linux raspberrypi 5.4.83-v7+ #1379 SMP Mon Dec 14 13:08:57 GMT 2020 armv7l GNU/Linux

starbasessd, 2021-02-01:
How did you install Docker?

Did you try just the base 'recommended' docker command line?

docker run --name="motioneye" \
    -p 8765:8765 \
    --hostname="motioneye" \
    -v /etc/localtime:/etc/localtime:ro \
    -v /etc/motioneye:/etc/motioneye \
    -v /var/lib/motioneye:/var/lib/motioneye \
    --restart="always" \
    --detach=true \
    ccrisan/motioneye:master-amd64

It picks up whatever time zone you are using on the Pi.
What port is 8999? It's not a motionEye port.

As an aside, there are some dependency issues that I haven't been able to resolve by upgrading motion in the docker image. Even upgrading both docker & ffmpeg, there are a lot of codec files that can't or won't upgrade. I've tried to do a Build version, and force the 4.3.2-1 version of motion, but it's still the same.
The way I accessed the container is
docker exec -it motioneye bash
with motioneye already running.

sukanth, 2021-02-01:
@starbasessd  yes, I did install docker using the recommended command-line method. I followed this blog post to install docker on raspberry pi  "https://pcmac.biz/docker-on-raspberry-pi/", please forget about the port.. I am using the right one.. 8765. I don't have issues with the timezone. I only wanted to upgrade the motion version. and this "motioneye:master-amd64" image that you mentioned I believe doesn't work on the raspberry pi since RPI is an arm based processor. 
So you were not able to upgrade to the latest version of motion using the image as well as the build version?

starbasessd, 2021-02-01:
That is correct.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-03-05:
No response in >30 days. If you are still having issues, please open a new issue.

sukanth, 2021-03-05:
> No response in >30 days. If you are still having issues, please open a new issue.

yes, I still am not able to upload to dropbox..

vlinnie, 2021-05-01:
I also have this problem with the docker version

kestena, 2022-01-07:
I found a way to do it under docker.

Go to this link from ccrisan:
https://github.com/ccrisan/motioneyeos/blob/master/package/motioneye/dropbox.keys

You will find the keys to Dropbox there.

Now go to your raspberry pi, and locate the uploadservices.py file. Some find it under /usr/local/lib/python2.7/dist-packages/motioneye/uploadservices.py, I personally found mine under docker since I run my version under docker.

If you use docker locate the file using `sudo find . -name uploadservices.py `.

The path will be something like docker/use/lib/overlay2/somenumbersandletters/usr/lib/etc.... 

I stopped the docker container first before attempting the changes in the uploadservices.py file. To do that just list the container under docker with `docker container ls`

Then copy the motioneye container (letters and number). Stop the container: `docker container stop [name of container]`

Nano to uploadservices.py, scroll until you find the Dropbox section and exchange the "dropbox_client_id_placeholder" with the first client_id from the Github link I pasted in the beginning. 

Do the same for the second setting there named: NOT_SO_SECRET something. Exchange the parameter with the second parameter from the Github link. Save with CTRL X, Yes, Enter. Restart the docker container with `docker container start [name of container]` and you are done here.

Go to the Motioneye settings, under Dropbox now it should work!

Hopefully I was able to help you.

Regards,
mislishsezahaker

sukanth, 2022-01-07:
> 

Thank you so much @kestena will try that out and let you know if that works

noeldum, 2022-10-01:
Same problem with dropbox "Test Service" hanging and not going anywhere.
This is on raspbian bullseye where motioneye was installed.

motionEye Version | 0.42.1
Motion Version | 4.3.2
OS Version | Raspbian 11

I checked the CLIENT_ID and CLIENT_NOT_SO_SECRET and they are the same as indicated in the link.

It works on the motioneyeos image on a pi but here on the installed version on top of Raspbian it does not work.

noeldum, 2022-10-01:
same thing with google drive actually. Everything just hangs.

MichaIng, 2022-10-02:
Please try the Python 3 based dev version which should have this fixed: https://github.com/motioneye-project/motioneye/tree/dev#installation

noeldum, 2022-10-02:
ok. I have flashed a new raspbian and re-installed motioneye as suggested.

Still running into troubles with dropbox.

I had to change the CLIENT_ID key as mentioned above. Not sure why this is not done at installation time.

Then I came across this error when doing "Test Service".

[image]

Looking at the  uploadservices.json file I did get my access_token and refresh_token populated. The upload does not seem to function though.

noeldum, 2022-10-02:
alright. upload working now. Was far behind for some reasons.

qoijjj, 2023-07-26:
@MichaIng this issue is present on the latest dev version. I'm not using docker and am running raspberrypios 11. This issue should be reopened.

qoijjj, 2023-07-27:
In the python2 branch, this placeholder is replaced here:

https://github.com/motioneye-project/motioneye/blob/python2/setup.py#L30

where is this done on the dev branch?

lucadjc, 2023-09-25:
> I found a way to do it under docker.
> 
> Go to this link from ccrisan: https://github.com/ccrisan/motioneyeos/blob/master/package/motioneye/dropbox.keys
> 
> You will find the keys to Dropbox there.
> 
> Now go to your raspberry pi, and locate the uploadservices.py file. Some find it under /usr/local/lib/python2.7/dist-packages/motioneye/uploadservices.py, I personally found mine under docker since I run my version under docker.
> 
> If you use docker locate the file using `sudo find . -name uploadservices.py `.
> 
> The path will be something like docker/use/lib/overlay2/somenumbersandletters/usr/lib/etc....
> 
> I stopped the docker container first before attempting the changes in the uploadservices.py file. To do that just list the container under docker with `docker container ls`
> 
> Then copy the motioneye container (letters and number). Stop the container: `docker container stop [name of container]`
> 
> Nano to uploadservices.py, scroll until you find the Dropbox section and exchange the "dropbox_client_id_placeholder" with the first client_id from the Github link I pasted in the beginning.
> 
> Do the same for the second setting there named: NOT_SO_SECRET something. Exchange the parameter with the second parameter from the Github link. Save with CTRL X, Yes, Enter. Restart the docker container with `docker container start [name of container]` and you are done here.
> 
> Go to the Motioneye settings, under Dropbox now it should work!
> 
> Hopefully I was able to help you.
> 
> Regards, kestena

@kestena, hi, I'm receiving the following error after the key handshaking. is it working to you?
 ERROR: dropbox: request failed: POST data should be bytes, an iterable of bytes, or a file object. It cannot be of type str.

wokawoka, 2025-03-13:
I'm also facing this issue
(running a motioneye LXC container on proxmox through https://community-scripts.github.io/ProxmoxVE/scripts?id=motioneye )

JamBalaya56562, 2026-11-13:
@MichaIng `Dropbox.CLIENT_ID` and `CLIENT_NOT_SO_SECRET` in `motioneye/uploadservices.py` are still `dropbox_client_id_placeholder` and `dropbox_client_secret_placeholder`, also in the 0.44.0 wheel on PyPI. *Obtain Key* therefore leads to Dropbox's error page (`unknown_client_id`, `Invalid client_id: "dropbox_client_id_placeholder"`), so Dropbox upload cannot be set up on a pip install, as reported here and in #171. Reopening, as nothing has changed.

The keys were removed from git in aefb88ff (2015). The python2 branch put them back at build time: its `setup.py` replaced the placeholders from an `extra/dropbox.keys` file that was not in git. Since the move to Python 3, nothing does that, and no workflow fills them in either. The UI has no field for an app key, so users are not meant to bring their own; the workaround passed around here and in #171 is to edit `uploadservices.py` by hand with keys from the motionEyeOS repository.

Options I can see:
1. Register a Dropbox app for the project and fill in its key and secret at release time, from repository secrets in `pypi_release.yml`.
2. Let users enter the app key and secret of their own Dropbox app in the upload settings, as for S3.
3. Remove the Dropbox upload service, as it cannot work in any release without one of the above. Other upload services remain, e.g. S3, WebDAV, SFTP and FTP.

Refreshing the access token uses the same keys, so this also matters for the refresh handling discussed in #3128. Which way fits the project is your call; I can prepare the code for any of them.

MichaIng, 2026-11-13:
Hmm, so it is similar to our Google OAuth2 client. Setting up a client ourselves saves users some steps, and allows us to pre-configure a sane name and icon, I guess. But we need a Dropbox account, and I deleted mine a long time ago 😄. Can create a new motionEye-only otherwise dummy account, of course.

However, we are not the motionEye server operators. Allowing users to register their own clients, which they do operate, makes arguably more sense. Ultimately we could allow both: Enter your own client, or use the builtin one. Shall the builtin account/client get lost again, an own client remains a native alternative.

Is there a reason to keep the client secret secret? It will be in plain text in the plain text Python script anyway. Worst that can happen is that people use that to register other applications as motionEye. Is there any problem with that? For Google uploads `CLIENT_NOT_SO_SECRET` is used.

@zagrim @Marijn0 does one of you have a Dropbox account, and would be willing to setup a motionEye client, to replace the placeholders, accepting that it may be freely (ab)used by anyone else for other applications? I guess there could be some statistics about how many Dropbox accounts it is attached to or so, which might be interesting for us, to have some idea of user numbers, and can be misleading if people mass-use public client credentials for other applications, for whichever reason this could be done 😄.

zagrim, 2026-11-14:
@MichaIng I don't think I've ever had a Dropbox account.

But wouldn't using a shared account mean that also all the files put there are shared (and inherently public since the client secret is not really a secret)? I have no idea how Dropbox operates, or Google Drive, in this matter, and maybe there is a twist here.

JamBalaya56562, 2026-11-14:
@zagrim The app key and secret only identify the app, not a storage account: each user still signs in to their own Dropbox and allows motionEye there (the OAuth code flow in `uploadservices.py`), and the token from that is stored in their camera settings. So the files go to each user's own Dropbox, and with the secret someone could only make their own app appear as motionEye, not reach anyone's files.

zagrim, 2026-11-15:
Ok, I see, thanks. Now this starts to make sense to me.

> Worst that can happen is that people use that to register other applications as motionEye. Is there any problem with that?

I'd guess there might be a way (in some use case or scenario that might not be too easy to execute) to coax a MotionEye user to allow fake-motionEye application to have access to their Dropbox account, but then again, couldn't the adversary anyway create a typosquatted or otherwise mimicked application to fool users 🤷 
Anyway, reading the answers for https://security.stackexchange.com/questions/138816/what-is-the-worst-that-can-happen-if-your-oauth-client-secret-is-leaked leads me into assuming there has to be some other vulneratibility in order for an attacker to be able to do some actually harmful things even if they had access to the OAuth client credentials.

I do see the benefit for users for providing a "motionEye" Dropbox app and its credentials for them (since creating an app for themselves is a sort of techy-mumbo-jumbo a non-technical user might feel scared of), but it still feels somehow "wrong" thing to do 😅

JamBalaya56562, 2026-11-15:
@zagrim @MichaIng Dropbox can do without a secret: its [OAuth guide](https://docs.dropboxapi.com/dropbox-api/docs/oauth) lists open source applications among those that cannot keep the app secret and recommends the code flow with PKCE for them. With PKCE only the app key goes into `uploadservices.py`. The code exchange sends a `code_verifier` instead of the secret, and a token refresh sends just the refresh token and the app key (the "PKCE refresh token request" example of [`/oauth2/token`](https://www.dropbox.com/developers/documentation/http/documentation#oauth2-token)). The app key is meant to be public, so nothing secret would be in the source, and a field for users' own app would need only the key.

If that is fine, I'll register the app, switch Dropbox to PKCE and test sign-in and a token refresh with it before opening the PR.

MichaIng, 2026-11-15:
> I'd guess there might be a way (in some use case or scenario that might not be too easy to execute) to coax a MotionEye user to allow fake-motionEye application to have access to their Dropbox [...]

Yeah exactly, anyone can create an OAuth2 application with any name and icon or whichever attributes it can have to mimic motionEye. Relevant is that it is registered through the exact one-time link generated in our backend, which ties the client users authorize via the link to their actual motionEye instance.

Generally, AFAICS, keeping an OAuth2 client application secret makes sense only for clients which are actually operating from or through a central infrastructure which belongs to the owner of the client software. If you add a Dropbox share to your Google Drive (if that is even possible), then it makes sense that Google, as operator of the server from which the Dropbox OAuth2 client requests are coming from, can use it exclusively, change or revoke it as needed. So the client identity serves as authenticity that the requests are actually coming from Google, and no one else can claim to be a Google client (can mimic, but the ID will be different). In our case, however, there is no single client operator, as everyone hosts its own motionEye. This sort of underlines my prior point, that users creating their own clients generally makes more sense, as then the client identity really means it is their own motionEye instance, and no other motionEye instance, that could damage to their data/privacy like any other client mimicked or not.

So all we do is saving users some steps by offering a public client, with some meaningful name and icon for them to find it easily in their Dropbox settings. This serves zero authenticity purpose, which is instead all coming from the way OAuth2 authentication works and is implemented our end, which makes it naturally hard for attackers. Latest motionEye also uses strict CSP rules, to mitigate XSS attacks, which are otherwise a way to compromise the server, grab inputs etc.

@JamBalaya56562
For Google uploads we already use a PKCE verifier. But we also still send the client secret and receive an access token along with the refresh token. So this seems to work slightly different, or do we just mix two methods together there? However, while it seems more reasonable for an open source application without central client host/network to not use a client secret at all, and also for the actual client instance to use a verifier in the backend to protect against stolen tokens, I do not see how these are related. If not secret is used/needed, then everyone can anyway use any client (ID)? Or can this be configured when creating the OAuth2 client at Dropbox, whether it requires a secret or not?

JamBalaya56562, 2026-11-15:
@MichaIng I tried it with a test app on a new Dropbox account. The app settings have a switch "Allow public clients (Implicit Grant & PKCE)", on by default for new apps. With it on, the code exchange with only `client_id`, `code` and `code_verifier`, a refresh with only `refresh_token` and `client_id`, and an upload all worked, no secret involved. With it off, the authorize link is refused with `pkce_disallowed`. So yes, it is configured per app.

On how they relate: anyone can use a public app key, as with the secret in our source today. What PKCE adds is that only the motionEye instance which created the authorize link can turn the code into tokens, so an intercepted code is useless. For Google we do the same, PKCE plus a secret that [Google itself says](https://developers.google.com/identity/protocols/oauth2#installed) is "obviously not treated as a secret" for installed apps, and [lists as optional](https://developers.google.com/identity/protocols/oauth2/native-app#exchange-authorization-code) for them.

MichaIng, 2026-11-15:
Let's wait a bit whether @ccrisan answers. If we can keep using the not so secret Dropbox client ID+secret from motionEye 0.42.1, it would be least change, also for users who might still upgrade, and have the old client registered. Adding PKCE however additionally makes sense, if Dropbox supports both the Google does.