motioneye-project/motioneye #1959: Adding a DVR running on port 15961

alijundi, 2020-12-01:
I have a DVR running mobile port on 15961 and it is perfectly running when I use the android MEye Pro HD. How do I add the 16 channels on my DVR to MotionEye. I really need to add the feeds to home assistant. I tried 

http://192.168.1.224:15961/
Empty reply from server

Also, this DVR does not support RTSP protocol for some reason.

Thanks alot!

starbasessd, 2020-12-01:
The DVR would have to supply 16 separate video streams on 16 separate
ports. Since you do not mention which DVR, we can't look it up to see if it
meets that spec.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

alijundi, 2020-12-01:
It is a chinese H264 DVR with one port for Mobile Access that 15961 and it works perfectly on android I attach a picture to prove.
[image]

alijundi, 2020-12-07:
My DVR only support mobile access!

starbasessd, 2020-12-07:
Probably not going to work for you, then.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2020-12-07 by alijundi