motioneye-project/motioneye #1960: pip install OOM

angadsingh, 2020-12-02:
`sudo pip install motioneye` oom's on raspberry pi zero w (with just 512 mb of ram)

kindly update the wiki at https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian and add this instruction:

on low memory boards like raspberry pi zero w use the `--no-cache-dir` with pip install to use a constant space algorithm to avoid OOM

starbasessd, 2020-12-02:
Will do. On my list.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-12-02:
Follow up:
created new RaspberryPiOS 2020-08-20 SCDard
added wpa_supplicant.conf & ssh to /boot
installed in RPiZeroW, and ran update & upgrade
no other packages installed
ran:
sudo su
pip -V
pip 18.1 from /usr/lib/python2.7/dist-packages/pip (python 2.7)
apt-get install ffmpeg libmariadb3 libpq5 libmicrohttpd12
wget https://github.com/Motion-Project/motion/releases/download/release-4.2.2/pi_buster_motion_4.2.2-1_armhf.deb
dpkg -i pi_buster_motion_4.2.2-1_armhf.deb
apt-get install python-pip python-dev libssl-dev libcurl4-openssl-dev libjpeg-dev libz-dev
pip install motioneye
     [no errors or OOM]

Before I change it, what did you have different, or do differently, so I can reproduce it before making the change in the instructions?

I then continued:
mkdir -p /etc/motioneye
 cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
mkdir -p /var/lib/motioneye
cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service
 systemctl daemon-reload
 systemctl enable motioneye
 systemctl start motioneye
Then confirmed I was able to connect to the motioneye @ http://[ip_address]:8765

starbasessd, 2020-12-02:
root@rp0-1-tesst:/home/pi# free
              total        used        free      shared  buff/cache   available
Mem:         442164      106852      108544        1824      226768      281308
Swap:        102396        2560       99836
root@rpi0-1-test:/home/pi# df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       7.0G  3.0G  3.8G  44% /
devtmpfs        184M     0  184M   0% /dev
tmpfs           216M   16K  216M   1% /dev/shm
tmpfs           216M  3.2M  213M   2% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           216M     0  216M   0% /sys/fs/cgroup
/dev/mmcblk0p1  253M   54M  199M  22% /boot
tmpfs            44M     0   44M   0% /run/user/1000

root@rpi0-1-test:/home/pi# uname -a
Linux rpi0-1-test 5.4.72+ #1356 Thu Oct 22 13:56:00 BST 2020 armv6l GNU/Linux

root@rpi0-1-test:/home/pi# cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

starbasessd, 2020-12-16:
Please respond.

felmoltor, 2020-12-27:
I had the same problem as @angadsingh when installing motioneye on a raspberry pi zero. 
The OOB kill happens when trying to install the pillow package. 
An alternative to install pillow was to execute the following:
`apt install python-pil`
And then, after the successful installation, ejecute the command:
`pip install motioneye`

starbasessd, 2020-12-27:
Which Raspberry Pi OS? Lite, Desktop, or Full? 
Any other software installed prior to following the instructions here:
https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29
What date version? 20201202 or earlier?
I was not able to reproduce on the 202012 or 202008.

felmoltor, 2020-12-27:
Hardware: Pi Zero W
OS: 2020-12-02-raspios-buster-armhf-lite

starbasessd, 2020-12-27:
As posted above, that is what I used RaspberryPiOS Lite, Dec 2 2020 from
RPiFoundation website, PiZeroWireless v1.1, and the only mod made before
running apt Update && apt upgrade -y was to add wpa_supplicant.conf and ssh
to /boot before first boot. I used the instructions for Ubuntu 20.04 and
Newer from the motionEye Wiki. No OOM errors anywhere.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-12-27:
Running it all again to confirm again.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-12-27:
PiZeroW v1.1, RaspberryPiOS 20201202, 16GB microSDCard, fresh install with Win32DiskImager.
Copy wpa_supplicant.conf and ssh.txt to /boot
Insert and power on
sudo su
apt update && apt upgrade -y
pip -V shows command not found
ditto pip2 -v
so you need to install pip2 via the instructions in the Ubuntu 20.04 or Newer instruction set.
I just went through the whole setup again, no OOM errors, no other errors.
I cannot reproduce the issue
Unless I can trigger the OOM error, there is no need to change. 
I will be modding the Debian based installs (Debian, Ubuntu, RaspberryPiOS) to all point to the Ubuntu 20.04 or newer instruction set since it would appear that they've all removed pip2 from the upstream repositories.

closed 2020-12-27 by starbasessd

jimis, 2021-06-03:
The OOM just happened to me on a freshly installed Raspberry Pi OS Lite.

> Copy wpa_supplicant.conf and ssh.txt to /boot
> Insert and power on
> sudo su
> apt update && apt upgrade -y

Up to this point we used same steps. Then I used the same instructions as listed at https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian  with the exception of installing Motion 4.3.2. 

Step 4 (pip install motioneye) ended with OOM kill, because my PiZeroW has 512MB RAM and 100MB default as per the default install.

Workarounds mentioned above work.