motioneye-project/motioneye #1961: Motioneye spams my camera folders with jpgs

AndreasHenn, 2020-12-04:
Hi,

I am using motioneye on a PI with 5 cameras. As a standard, I configured video devise, movies with H.264 (mp4), motion triggered and motion detection with 10frames before.

Now I can see, that motioneeye is spamming my folders for the videos with so many jpg, that I am not able to select them with one rm command (not even with 10, I need to programatic solve that).

On the one side, that is massivly droping my disc throughput, but it is also filling my data store so that after short time (~2month which is not much time for that :-) ), motion eye stops to work as there is no storage any more.

Any hint on that?
Kind regards
Andreas

starbasessd, 2020-12-04:
Add a sub-folder to the filename:
Normal:  %Y-%m-%d/%H-%M-%S
Sub-folder:  %v/%Y-%m-%d/%H-%M-%S
or %Y-%m-%d/%v/%H-%M-%S
%v is 'incident number' and increments after each motion detection event.
You can also reduce the number of jpg images by using "Motion Triggered (one snapshot)"

starbasessd, 2020-12-04:
You can also use an additional sub-folder to separate the movies from stills:
%Y-%m-%d/%H-%M-%S/jpgs
%Y-%m-%d/%H-%M-%S /moviest
or
%Y-%m-%d/jpgs/%H-%M-%S
%Y-%m-%d/movies/%H-%M-%S
or 
jpgs/%Y-%m-%d/%H-%M-%S
movies/%Y-%m-%d/%H-%M-%S
ad nauseum...

AndreasHenn, 2020-12-04:
Hi,

thanks for coming back so quick to me with your recommendations.

I understand the idea to move the files to a different location, but I would prefer to completely get rid of them.

I would prefer to just have the movies reserved for one week.
[image]

In addition I do not exactly know how to configure the location of the files as I just configure the Movie location.

I do currently have many millions of pics in my folders for 5 cameras, which is really slowing down the disc performance and only few movies are written (maybe caused by that).

Can you please advise again.

Thanks in advance
Andreas

starbasessd, 2020-12-04:
In the title bar for Still Images, on the left is 'ON', toggle to 'OFF'

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

AndreasHenn, 2020-12-04:
it has never been to ON, therefor quite strange

starbasessd, 2020-12-04:
Which camera(s) are showing in the jpegs? the 'images' setting is for each
individual camera, so if you see camera 2,3,4 &5 but not 1, then you only
turned it off for 1

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

AndreasHenn, 2020-12-04:
configuration is done same for all cameras

starbasessd, 2020-12-04:
Which camera(s) are spamming the folder?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

AndreasHenn, 2020-12-04:
nearly all... now I created a script to automatically delete them every hour

starbasessd, 2021-03-05:
If you are still having issues, please open a new issue.

closed 2021-03-05 by starbasessd