motioneye-project/motioneye #1962: New Raspberry pi installation - syntax error deep in asyncio

ctgreybeard, 2020-12-06:
I followed the wiki to install motioneye on my Pi 4. It all installed fine but there seems to be a python 2 compatability problem deep in asyncio which tornado is calling.  Here what I get:

```
-- Logs begin at Thu 2019-02-14 05:11:59 EST, end at Sat 2020-12-05 22:25:29 EST. --                                    │··
Dec 05 22:20:36 playpi systemd[1]: Started motionEye Server.                                                            │··
Dec 05 22:20:36 playpi meyectl[15217]: Traceback (most recent call last):                                               │··
Dec 05 22:20:36 playpi meyectl[15217]:   File "/usr/local/bin/meyectl", line 10, in <module>                            │··
Dec 05 22:20:36 playpi meyectl[15217]:     sys.exit(main())                                                             │··
Dec 05 22:20:36 playpi meyectl[15217]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/meyectl.py", line 254, i│··
Dec 05 22:20:36 playpi meyectl[15217]:     import server                                                                │··
Dec 05 22:20:36 playpi meyectl[15217]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 28, in │··
Dec 05 22:20:36 playpi meyectl[15217]:     from tornado.ioloop import IOLoop                                            │··
Dec 05 22:20:36 playpi meyectl[15217]:   File "/usr/local/lib/python2.7/dist-packages/tornado/ioloop.py", line 52, in <m│··
Dec 05 22:20:36 playpi meyectl[15217]:     from tornado.concurrent import Future, is_future, chain_future, future_set_ex│··
Dec 05 22:20:36 playpi meyectl[15217]:   File "/usr/local/lib/python2.7/dist-packages/tornado/concurrent.py", line 48, i│··
Dec 05 22:20:36 playpi meyectl[15217]:     import asyncio                                                               │··
Dec 05 22:20:36 playpi meyectl[15217]:   File "/usr/local/lib/python2.7/dist-packages/asyncio/__init__.py", line 9, in <│··
Dec 05 22:20:36 playpi meyectl[15217]:     from . import selectors                                                      │··
Dec 05 22:20:36 playpi meyectl[15217]:   File "/usr/local/lib/python2.7/dist-packages/asyncio/selectors.py", line 39    │··
Dec 05 22:20:36 playpi meyectl[15217]:     "{!r}".format(fileobj)) from None                                            │··
Dec 05 22:20:36 playpi meyectl[15217]:                                ^                                                 │··
Dec 05 22:20:36 playpi meyectl[15217]: SyntaxError: invalid syntax                                                      │··
Dec 05 22:20:36 playpi systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE                │··
Dec 05 22:20:36 playpi systemd[1]: motioneye.service: Failed with result 'exit-code'.
```

`raise ... from ...` is python3 syntax.

Is there a solution?

starbasessd, 2020-12-06:
Which set of instructions did you use?
Which RPiOS did you use? 2020-08-20 or 2020-12-02?
If 12-02, use the Ubuntu 20.04 0r later instructions, please.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ctgreybeard, 2020-12-06:
Thanks for the instant response, Ken!

I'm running Raspian 10 (buster), apt is fully up to date.

I installed using https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian including installing 4.2 of motion.

Bill

starbasessd, 2020-12-06:
There may be an issue with the upstream installs of python, since all
official support ends 12/31/2020. Please try the Ubuntu 20.04 and after
instructions wich uses a different method to install pip2, and the
instructions force 2.7 python apps.
I am testing right now, on a Pi4-4GB...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ctgreybeard, 2020-12-06:
I should mention too that by "new installation" I meant that motioneye is the "new" part, the pi has been running for quite a while.

As I am using this for "playtime" I am also very willing to test. I'm fluent in "computer" ...

starbasessd, 2020-12-06:
<grin>

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ctgreybeard, 2020-12-06:
Went through the Ubuntu install and only pip was changed. Still fails in the selectors module.

Strange that the 2.7 package contains python3 syntax. But 2.7 is "dead" so I guess you can't complain.

Looking at the asyncio package under 2.7 (`/usr/local/lib/python2.7/dist-packages/asyncio`) it was installed May 11 this year. There are several instances of `from None` but only in two modules (`selectors.py` and `unix_events.py`)

I'm considering "fixing" those two to see what happens.

ctgreybeard, 2020-12-06:
The issue is described here: https://github.com/python/asyncio/issues/386

ctgreybeard, 2020-12-06:
Interesting. There is no asyncio installed on my motioneyeOS systems. I have two PiZeros with that installed.

starbasessd, 2020-12-06:
How do you trigger your error?
I have a new fresh install of Raspberry Pi OS and followed my installation
instructions at
https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ctgreybeard, 2020-12-06:
It fails when it runs under systemctl. I can also trigger it just by loading python2 and `import asyncio`

The error log I posted was from journalctl

starbasessd, 2020-12-06:
Can you post the result of
sudo systemctl status motioneye
please?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ctgreybeard, 2020-12-06:
```
● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2020-12-06 11:09:52 EST; 1h 52min ago
  Process: 18678 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=1/FAILURE)
 Main PID: 18678 (code=exited, status=1/FAILURE)

Dec 06 11:09:52 playpi meyectl[18678]:   File "/usr/local/lib/python2.7/dist-packages/tornado/concurrent.py", line 48, in <module>
Dec 06 11:09:52 playpi meyectl[18678]:     import asyncio
Dec 06 11:09:52 playpi meyectl[18678]:   File "/usr/local/lib/python2.7/dist-packages/asyncio/__init__.py", line 9, in <module>
Dec 06 11:09:52 playpi meyectl[18678]:     from . import selectors
Dec 06 11:09:52 playpi meyectl[18678]:   File "/usr/local/lib/python2.7/dist-packages/asyncio/selectors.py", line 39
Dec 06 11:09:52 playpi meyectl[18678]:     "{!r}".format(fileobj)) from None
Dec 06 11:09:52 playpi meyectl[18678]:                                ^
Dec 06 11:09:52 playpi meyectl[18678]: SyntaxError: invalid syntax
Dec 06 11:09:52 playpi systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Dec 06 11:09:52 playpi systemd[1]: motioneye.service: Failed with result 'exit-code'.
```

starbasessd, 2020-12-06:
It indicates that it never correctly installed. Please use the Ubuntu
(20.04 and newer) instructions from the beginning.
Be sure to be in root (sudo su or similar)...
No need to re-image.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ctgreybeard, 2020-12-06:
Found it!!

*Because* I had `asyncio` installed (from another project) tornado uses it. Bad ...

Uninstall `asyncio` and it runs.

starbasessd, 2020-12-06:
Cool beans.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2020-12-06 by ctgreybeard