motioneye-project/motioneye #1963: Missing frames in recordings

sdevrieze, 2020-12-09:
I am using motioneeye in a shop environment. This means you sometimes get a lot of motion during during the peak times of the day, whilst at other times there is no motion during several minutes. To avoid unnecessary network traffic, all Pi3/Pi4 cameras are running motioneye and recording on the microsd card. I am using 256GB high endurance cards and a cronjob that automatically deletes the oldest day directory with recordings as soon as empty space is running low. To save space and to make it easier to follow up on shoplifters, recordings are motion triggered and maximum 600 seconds long.

All seems to be working fine at first sight, but when I go look a bit deeper there is a serious problem in the recordings. Often multiple frames are not being saved in the recordings during several seconds. When watching the downloaded mp4 files in VLC, t can even jump from second 10 to second 30, meaning a timeframe of 20 seconds is not covered by recordings. This is a big deal as in even a few seconds a lot can be stolen.

Any idea what is causing this and how to avoid lost frames?

I have seen this issue September 2019 on a Pi3 camera running MotionEyeOs. Today I am watching a day of recordings and I am currently experiencing the same issue on a camera running MotionEye on Raspbian on a Pi4.

sdevrieze, 2020-12-09:
Just an example: I am watching a recording that starts with a frame with the time 10:44:12 and ends with a frame timed 10:53:26. This is a timeframe of 9 minutes and 14 seconds. However, according to VLC the total length of the video file is only 7 minutes and 2 seconds. So 2 minutes and 12 seconds was not recorded for whatever reason.

starbasessd, 2020-12-09:
You didn't anwer any of the questions needed to help you, including versions of motion & motionEye, which OS and version, Hardware used (Pi or other, cameras, etc) no logs included, etc.

sdevrieze, 2020-12-09:
I do not recall the version that I was using last year, but the recordings were done on August 30th 2019. That means it was a MotionEyeOs version from before that date. If I remember correctly, I installed the software on that camera in June already. It was a Raspberry Pi3 B+ with a camera using the camera interface.

The recent case is on a Raspberry Pi4 B 2GB:
motionEye Version | 0.42.1
Motion Version | 4.3.1
OS Version | Raspbian 10

What logs do you need?

starbasessd, 2020-12-09:
/var/log/motion.log for the timeframe involved
/var/log/dmesg for the timeframe involved
/var/log/messages for the timeframe involved
If there is a motioneye log, include it, too.
If not, It might be helpful to add a function to record the temperature and
voltages while running, to see if there are any issues.
If you have not rebooted the PI since the issue, can you run in either a
SSH session, or in the console
vcgencmd get_throttled [enter]
 and post the result.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sdevrieze, 2020-12-09:
[messages.log](https://github.com/ccrisan/motioneye/files/5667145/messages.log)
[motion.log](https://github.com/ccrisan/motioneye/files/5667136/motion.log)

There is no dmesg or motioneye log.

Output is: throttled=0x0

starbasessd, 2020-12-09:
How about syslog?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sdevrieze, 2020-12-09:
[syslog.1.log](https://github.com/ccrisan/motioneye/files/5667252/syslog.1.log)

starbasessd, 2020-12-09:
Timeframe of the dropped frames?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sdevrieze, 2020-12-09:
For instance: 10:47:31 jumps immediately to 10:47:48.

starbasessd, 2020-12-09:
I am not seeing anything obvious in the log.
I have a setup
motionEye Version 0.42.1
Motion Version 4.3.1
OS Version Ubuntu 20.04
I do not have that issue.
What framerate and resolution are you using for the CSI camera?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sdevrieze, 2020-12-09:
resolution: 1280x960
video rotation: 180°
framerate: 10

Text overlay is enabled, video streaming is enabled with motion optimization, movie format is H.264/OMX with quality 75%, motion detection has auto noise detection and show frame changes enabled.

starbasessd, 2020-12-09:
Can you try a non-omx format?
This is going to be a @jasaw or @ccrisan issue...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sdevrieze, 2020-12-10:
I can try, but that means higher CPU usage, isn't it?

starbasessd, 2021-03-05:
If you are still having issues, please open a new issue.

closed 2021-03-05 by starbasessd

sdevrieze, 2021-03-05:
AFAIK this is not yet fixed. Has there been a new release or a patch that I should test?