motioneye-project/motioneye #1964: Action Buttons on MotionEyeOS not working (An error occurred. Refreshing is recommended)

zapp-bit, 2020-12-10:
Hi there. 

As the title states, the action buttons do not work, when pressed, I get the error "An error occurred. Refreshing is recommended"

I can confirm the script works, when executing it from command line, it works a charm. The issue seems to be with how the web gui is trying to execute it. I have attached the relevant log information below.

`2020-12-10 18:18:51: [motioneye]    ERROR: [Errno 8] Exec format error
Traceback (most recent call last):
  File "usr/lib/python2.7/site-packages/tornado/web.py", line 1548, in _stack_context_handle_exception
  File "usr/lib/python2.7/site-packages/tornado/web.py", line 1756, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 1685, in post
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 1688, in run_command_bg
  File "usr/lib/python2.7/subprocess.py", line 394, in __init__
  File "usr/lib/python2.7/subprocess.py", line 1047, in _execute_child
OSError: [Errno 8] Exec format error
2020-12-10 18:18:51: [motioneye]    ERROR: 500 POST /action/1/alarm_on/?_=1607584731568&_username=admin&_signature=e13c31ce11861bcdaf5dfabc5fcf05023fa2ab80 (192.168.1.50) 20.30ms
`

The script is a copy/paste of your basic script to turn a light on and off, and i am using a raspberry pi 3 b+

Please tell me if you know how to fix this. 
Thanks

starbasessd, 2020-12-10:
MotionEyeOS, or motionEye?
Which install instructions or which version?
What underlying OS if motionEye?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

zapp-bit, 2020-12-10:
It is most certainly MotionEyeOS
I used the install insructions the creator has posted here - https://github.com/ccrisan/motioneyeos/wiki/Installation
Used Balena Etcher to write it to the SD card.

These instructions to make the action buttons files - https://github.com/ccrisan/motioneyeos/wiki/Action-Buttons

starbasessd, 2020-12-10:
Which version of motionEyeOS? 20190911, 20200203, 20200606, dev20201026?
(found in Settings, General or as part of the name of the file downloaded)
This is the motionEye repository, there is one specifically for motionEyeOS
at
https://github.com/ccrisan/motioneyeos/wiki
which is specifically focused on motionEyeOS, ThingOS, and related issues
to the fact that there are many features and file systems locked down that
aren't in motionEye on other OSs.
If you open your issue on the motionEyeOS, please post your script, and
where you placed it.
If your script is python3 based, try specifying python3 calls, and not
python calls.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

zapp-bit, 2020-12-10:
Sorry about that. Are you able to move this to the motionEyeOS repo? or do I have to post it there?

here are the additional details

motionEye Version | 0.42.1
Motion Version | 4.3.1
OS Version | motionEyeOS dev20201026

starbasessd, 2020-12-10:
I can copy/paste but you wouldn't be notified

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

zapp-bit, 2020-12-10:
No worries. I posted it in the correct place, hopefully it helps to sort it out. 

For anyone who is interested, follow it here - https://github.com/ccrisan/motioneyeos/issues/2603

closed 2020-12-10 by starbasessd