motioneye-project/motioneye #1965: Frequency of motion triggered still images.

basta, 2020-12-11:
Hi.
If I understand it correctly, when saving motion triggered still images, it takes still image every second of motion. Is there (or can there be) any way of setting this frequency? I would like (for example) a picture of the motion every 10 seconds. The pictures are (and must be) in 4k so I do this because of space requirements.
If there is a way to code this please point me in a direction and I could do it and submit it.
Thanks

starbasessd, 2020-12-11:
If I am not mistaken, you may need to look in motion, as motionEye is a front end making the settings available in a more user-friendly way.
That said:
In motion configuration:
On Event, On Picture, On Motion commands, you might be able to put in a script with a 'sleep 10' pause, or something similar. You can add that script in the Extra Motion Options per Video Device.

starbasessd, 2020-12-11:
A list of configuration options is available at
https://motion-project.github.io/motion_config.html
Scroll down to  Configuration Options-Listed Alphabetically
Then click the link for the option in the last column to the currently available setting description

zagrim, 2020-12-11:
Yeah, I don't think that kind of thing is possible without some script magic, since Motion currently offers only three modes for motion triggered still images: first frame, "best" frame (largest amount of motion) and the "usual selection" which I guess means either every frame passed through motion detection or some hardcoded subset of all of the frames. 
https://motion-project.github.io/motion_config.html#picture_output

basta, 2021-02-14:
My final solution was running a python script on the file, which would remove it, if the last picture saved was less than 10s ago. Using a hook in the configuration.

JamBalaya56562, 2026-11-10:
Closing, as this was answered: Motion offers only first, best or every detected frame for motion-triggered pictures (`picture_output`), so a fixed interval needs a script on the picture hook, which is how the reporter solved it.

closed 2026-11-10 by JamBalaya56562