motioneye-project/motioneye #1966: Watch movie while recording

Pingbo, 2020-12-11:
Hi all,

I currently try to move away from my Synology NVR solution and found Motioneye pretty nice.

Only Point what i'm missing is to watch/Forward/rewind the movie while it's recording.

So in my Synology i can rewind the Stream to one or two minutes before Live. As i'm using one of my cams as pet Cam and want to Check what happened some minutes before.

Is there a possibility to do the same in Motioneye or perhaps implement a similiar Feature?

starbasessd, 2020-12-11:
Not currently implemented, you could shorten recording lenth to 30 seconds
or 1 minute, reduce motion gap to 1 second, etc.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Pingbo, 2020-12-14:
Thanks for the Information.
Sadly it not really suites into my needs.
Because is currently recording 24/7 and for the case i hear something strange at night. I check the cam what happend the last 10 Minutes. As my dogo is sleeping in another room at night.

Would be nice if you can see this like an feature request :)
For now i will use the Synology solution, but have to switch in the future. So will follow motioneye for sure 👍

TheMrRandomDude, 2021-01-29:
Also bumping this.

I record 24/7 too in case of burglars, and I'd really like to have an option to have motion activated video saved specifically and continous videos specifically. (Can't have both - which I would like).

I had my 3 cameras running 1 month ago and a dog ran into our backyard and caused a lot of noise. Only one camera's motion trigger activated, so naturally the other 2 didn't record anything. Imagine if it was a burglar. Evidence lost.

That's the first thing I want. Something to log motion events and to record them specifically, while also recording continously.

And the second thing is directly related to this issue post, I'd also like to be able to view all my 3 cameras like 3 minutes in the past, just like how @Pingbo explained it.

For example, I just heard some noise and I want to check all cameras 30s ago till now in fast rewind mode.

---

I think those are the most important things so far. It could maybe be integrated in a way like, I have continous recordings set to 30 minutes, and I noticed only mp4 files are playable by the WebUI, and they can't be played till they're finished recording because of moov atom issue.

On the other hand, formats like MKV can be played, but they record huge videos, almost 15Mbps compared to my Pass-through 0.8Mbps video. Now, since MKV is a container that will gladly accept any vid, I was thinking, maybe we could Pass-through raw video frames to MKV and use it's format for playback?

That solves the 2nd issue,m that is - @Pingbo's issue.

---

Regarding the 1st issue, I think it's also a simple solution. I'll write a script that will be ran and will log motion trigger events into a csv file. It should be very simple to actually integrate in motioneyeos to have separate recordings too. I'm just not good enough of a coder unfortunately.

And something with a nice UI would be much better than csv too! 

Thanks :)

starbasessd, 2021-01-29:
Easy to set up a hub machine, that acts as a file server. Cameras
individually save continuously to the local machine or to the file server
hub, and motion detect on the hub records the motion detect video there.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

TheMrRandomDude, 2021-01-29:
@starbasessd Oh that's a great idea! I already run Motioneye in a docker container, so I could easily do it. What about the feature Pingbo requested too though? Do you have any ideas about rewinding? That would be really fantastic to have.

starbasessd, 2021-01-29:
Not currently possible, BUT could set up another hub for that purpose,
with, say, 3 minute continuous recording, (because not all video formats
allow read while write), or the hub could use a format that does, but
discard after 24 hours...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?