motioneye-project/motioneye #1967: Leave fullscreen inconsistent via UI / Keyboard

lhartmann, 2020-12-12:
Leaving full-screen camera view has inconsistent behaviour:
- When triggered clicking on the "full-screen" icon it returns to the main view.
- When triggered by ESC / F11 the browser leaves full-screen,  but the layout is not updated and still requires clicking UI icons.

If this is by choice then never mind, but I find it annoying. Adding the code below to `$(document).ready()` on `/motioneye/static/js/main.js` should make the behaviour more consistent.
```
    $(document).addEventListener('fullscreenchange', function() {
        if (!$(document).fullscreenElement)
            doExitFullScreenCamera();
    });
```

P.S.: Sorry for not sending this as a PR, but I just don't know enough python to even test-run the code, so I could not even spell-check this.

zagrim, 2020-12-12:
I'm able to reproduce this, and I don't think it's by design, but rather due to the support for that event being still rather new (well, some years now, but ME is way older and so that was not possible to do in the initial implementation).

I can also try to produce a PR for this based on what you provided (thanks for that!).

zagrim, 2020-12-12:
I've not before really used Fullscreen API, but I guess what you @lhartmann provided is pretty much ok. Older browsers will not support that, but then again they should not also provide fullscreenchange events and thus that should be safe. I'll try to test that myself with IE11.

zagrim, 2020-12-13:
Let's have the more conceptual discussion on the single-camera vs. full-screen modes here instead of the PR.

I did some thinking and experimenting both with Youtube (theathre mode vs full-screen) and MotionEye. What is the essential difference between single-camera view and full-screen mode in ME @ccrisan? I fail to note any (except that in the former one can access browser menus etc).
With a small screen, high number of cameras and/or mobile use I recognise the value of having just one camera shown, but I didn't find change in the available functionality in ME in the single-camera mode compared to full-screen mode. They are starting to look like redundance to me, and cause some small glitches in the UX. For example, when concentrating to a single camera by entering full-screen mode, then opening the movies browser, there is a shortcut key to close the browser by pressing Esc, but as I'm in the full-screen mode, I need to press Esc twice - first to exit full-screen mode and then to close the browser. Not really a big deal, but having those two modes interconnected like they are is somewhat confusing. Maybe that's why I hadn't never really recognised that there are two separate modes available.
Anyway, I'm not saying that we should get rid of either of them, I'm just curious to know the rationale behind the current design.

ccrisan, 2020-12-13:
> What is the essential difference between single-camera view and full-screen mode in ME.

I often find myself wanting to watch a particular camera "in the background", meaning that I have one of my tabs/windows opened with one camera and not in full-screen. If it were in full-screen, I wouldn't be able to leave it and get back to it a bit later, considering it automatically exited full-screen.

It's not a big deal, but if we want to change something in this direction, the proposed PR is not the best solution IMO, for the reasons stated above. Having a separate option for this mode, like YT's "theater mode" sounds good, but I honestly don't have the time these days to implement it.

zagrim, 2020-12-13:
Well, IMO that **is** a valid reason to have both full-screen and single-cam view. If something like the YT "theatre mode" is an acceptable addition, I'll see if I am able to craft the necessary stuff to amend that PR (that will take at least a week or two). I've not yet looked at the code to see if that kind of thing would be easy to add, though.

zagrim, 2021-01-09:
I've been on and off from working with this, but I did manage to get a decent (IMO) solution which has both full-screen mode and single-cam mode, with the transitions between those and the normal multi-cam mode working nicely. In the process I added a new icon to the camera toolbar (2nd from the left) based on the icon design used in e.g. Youtube:
[image]

However, I started wondering if it would be more intuitive for the user if instead of the full-screen and single-cam buttons working like toggles (just like full-screen mode works currently) the buttons should depict transition to another mode. My thought being, that it might not be most intuitive for the user to click on "single-cam" button to get back to multi-cam mode.
Let me explain by example:
- in multi-cam mode the buttons would be like shown above
- in sinle-cam mode the "single-cam" button would change to something depicting the normal multi-cam mode (perhaps 2 or more rectangles)
- in full-screen mode the "full-screen" button would change to "multi-cam" button
(so basically the buttons would be like a radio-button group of three, with the activated button always being hidden)

Does that sound like a good idea, or would it just confuse some users?

Another way to clear the connection between the current mode and the button perhaps would be to somehow highlight the button which has activated the currently shown mode. So, after entering full-screen mode the left-most button would be highlighted, and after entering single-cam mode the button 2nd from the left would be highlighted.

And actually, perhaps the single-cam button should not be shown at all if the setup only has one camera :thinking:

starbasessd, 2021-01-09:
How about: it confuses regular users and help desk people?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

zagrim, 2021-01-10:
@starbasessd which of the three ways of working I mentioned in my comment did you refer to? :smiley: 

Let me re-cap:
The first way of the buttons working I mentioned, with buttons the same always (same icon, same tooltip) and acting as toggles, is the way full-screen button currently works (actually the tooltip _is_ "toggle full-screen mode" so there the logic is made clear). However, when adding a second "mode" button IMO it becomes confusing since there is no visual cue of whether a button is going to set the mode described by the icon and tooltip or restore the "normal" mode (which I'm calling multi-cam mode).

Then I drafted two possible new ways of working for the mode buttons as a solution for this ambiguity:

A) Make the mode button icons (and perhaps also their tooltips) change to show to which mode the UI would switch upon clicking on the button.

B) Add highlighting to the mode buttons so that the "current" mode would also be visible with the corresponding button being highlighted (e.g. the buttons would show that "this mode is now ON").

I'd of course like this to be as clear to both users and help desk people, but not being that competent with UX stuff I need some good second opinions on the solution. So, please comment on the options and/or propose a better way to solve this - I'm open for improvements.

starbasessd, 2021-01-10:
I'd like to see something like
Take to full screen:
[image]
Take back to normal:
[image]
Maybe a Next Camera (when full screen)
[image]

zagrim, 2021-01-10:
Thanks for the input and taking the effort to put in it mock-ups.

Did you notice, however, that there is a use case also for the single camera mode (without browser full-screen)? That's the reason for the need to have some (clear) way to also be able to have just a single camera shown without first putting the browser full-screen, and that takes us to the added icon shown in my earlier comment. And if it's the single-cam mode icon I came up with that is confusing, feel free to suggest a better one.

starbasessd, 2021-01-10:
By single camera view are you referring to http://[ipaddress]:8081 view?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

zagrim, 2021-01-10:
I'm referring with "single-cam mode" to the view where you end up after exiting full-screen mode by hitting Esc key (as described in the first comment in this issue). Please take a look at the 1st, 4th and 5th comment to get the reasons which led me to try to implement a change.

starbasessd, 2021-01-10:
OK, I don't use the keyboard when dealing with the WebGUI, so never noticed
that issue before. I still vote for the Full Screen, exit from full screen,
tab to next camera icons. When I used a Pro system back in my Casino IT
days, we parked the keyboard behind the monitors, and just moused
everything, so that's where I developed my habit at.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

lhartmann, 2021-01-12:
IMHO single/muti camera and fullscreen/windowed could be two separate toggles. MotionEye should go fulscreen or not via body.requestFullScreen(), and cameras should go single view or not via CSS.

The use case would be to get to fullscreen, multi cameras on touch devices, where we can't use F11.

Also the most frequent need during use (not setup) is probably zooming into a camera, so toggling single/multi camera views could be the default on-click action. Only fullscreen would need a toggle then.

zagrim, 2022-03-13:
I've now pushed update to the linked PR. Now it works so that one can enter full-screen mode from both multi-camera view and single-camera view, and it should return back to the previous view when exiting full-screen using either the browser controls or the camera toolbar button. I think this is consistent from the POV of navigation (exit from full-screen returns you to whatever you were at when entering full-screen). There's a gap with the logic though: If one has entered full-screen via single-camera view, both full-screen button (=exit from full-screen) and single-camera view buttons change the view to single-camera view.

Another UX thing that might be a bit off is the meaning of the camera top bar buttons in any given situation. Now the button icons refer to the mode they transfer the view when starting from the initial multi-camera view, but is it confusing to return to some-other view from full-screen using a button depicting "full-screen"? Youtube (which has been used as a reference for some reason here) uses different icon to return from full-screen, I just personally don't find that icon very intuitive (not to mention it resembles WW2 iron cross of Germany).

From those two above points I arrive at this question:
Would it be better if the buttons were changed so that from full-screen it would always be possible to go to both single-camera view and multi-camera view using the buttons (still maintaining the current idea of "return to previous view" when exiting full-screen with the browser controls)? This would mean swapping the full-screen button with a multi-camera view button when in full-screen mode.

zagrim, 2022-03-13:
> Also the most frequent need during use (not setup) is probably zooming into a camera, so toggling single/multi camera views could be the default on-click action. Only fullscreen would need a toggle then.

This would perhaps be ok for mobile use, but it would still mean one tap/click more to access saved images and videos, which is something users might be doing quite a lot, so I didn't go this route.

MichaIng, 2022-03-13:
> so toggling single/multi camera views could be the default on-click action

Makes sense to discuss this in a dedicated issue/PR, so for now only a single cam view is added without changing any behaviour that is not directly related.

> Would it be better if the buttons were changed so that from full-screen it would always be possible to go to both single-camera view and multi-camera view using the buttons

Probably it makes sense to always show the two buttons which are not the current view mode. Alternatively the active view button could be shown greyed or otherwise highlighted or with different colour, easy to do via CSS. But space is precious, so I'd personally vote for simply hiding buttons which wouldn't do anything anyway.

MichaIng, 2022-04-12:
Merged into `dev`.

PR #1968 (2022-04-12): Add support for fullscreenchange event to exit full screen mode
Closes #1967.
closed 2022-04-12 by MichaIng