motioneye-project/motioneye #1970: RTSP live video works only with last configured camera

Damian411, 2020-12-13:
Hello, I am new with the great project motioneye, but I have a issue with the live video in the GUI and the steaming URL.

motionEye Version: 0.42.1

Motion Version: 4.2.2

OS Version: Ubuntu 20.10

Installation with guide: 

https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29

Scenario:

1) After the first time configuration of my first RTSP camera I see a live video in the GUI and streaming URL.

2) RTSP of camera stops after some time --> My battery camera (Eufy) has an own motion detection and sends only a RTSP steram if it detects a motion to save battery.

3) The live video become gray "Unable to open video device"

4) Then I configure my second camera with the same procedure.

5) After a motion detection at my first camera it restarts the stream over RTSP

6) Motioneye detects a motion on the first camera and the border of the live video in the motioneye GUI become red, but the live video still leaves gray. Also the streaming URL still leaves gray. --> With VLC I see in this case a live video on the RTSP stream of the first camera.

7) BUT: Motioneye records and saves the movie. I can download and see the full record.

8) The live video of the second (and last configured) camera works fine in the GUI.

9) If I delete and configure the first camera once again, then it works, but not the second one anymore. So only the live video of the last configured camera is working.

Expected behavior: Motioneye obviously detects that the RTSP stream reconnects and records it, but the live video does not work anymore. I assume that Motioneye should show the live video in the GUI after the RTSP stream reconnects. I have the same behavior with Edge, Chrome and Firefox. If I click on the gray video I see the value of "frames per seconds (fps)", it changes every second, so the video refresh in the browser works. But the time in the bottom right corner of the live video leaves the same and is not refreshed.

Is this a bug or make I here a mistake?

starbasessd, 2020-12-13:
If you turn off the power save feature of the camera(s) (for testing
purposes) do you have both working? Does either live feed go gray screen
after a time?
There are several watchdog functions.
Are you running on a Pi or PC?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Damian411, 2020-12-13:
Deactivating of the power save feature is not possible, no option to do that.

Both live feeds go gray, because both cameras stops their RTSP streams. But only the live feed of the last configured camera comes back. The other still leaves gray.

It runs on an Intel NUC with still enough performance reserves.

starbasessd, 2020-12-13:
Are you able to SSH into the NUC?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Damian411, 2020-12-13:
Yes, I have full access.
Do you need an output?

starbasessd, 2020-12-13:
If both cameras are set up, and one or both are showing gray screen, in the
WebGUI, in a separate window ssh in and enter:
sudo systemctl restart motioneye [enter]
and tell me the result.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Damian411, 2020-12-13:
Sames behavior with one difference. Now both cameras still leaves gray, but the recording still works.

starbasessd, 2020-12-13:
Interesting. I'll leave this here to see if anyone else has ideas, but it
may be a bad interaction between your cameras with the 'power save'
function, and the ability for motioneye to reset live video.
Maybe @jasaw or @ccrisan have other ideas.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Damian411, 2020-12-13:
When I restart motioneye while the RTSP stream of one camera is active, then the live feed comes back immediately. 
After stopping and restarting of the RSTP stream the live feed does not come back once again.

So it looks like motioneye does not reactivate the live feed after an interruption of the RSTP stream.

starbasessd, 2020-12-13:
Any way to force the livestream from / on the camera?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Damian411, 2020-12-13:
I can force it for a limited time, normally maximum 60s. There are no other option and would not make sense for a normal useage, because the battery would be too fast empty.

starbasessd, 2020-12-13:
If it were a command, and you want to see the live stream, it could be
handy to force the camera, then reset the motioneye livestream view...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Damian411, 2020-12-14:
I could try it. I can configure a script like "if detect a motion then restart motioneye". In this case I would have a few seconds delay till I can see a live feed.
Better would be a script like "if RTSP is back, then restart motioneye", but I do not have an idea, how I can check the RTSP stream with an script.

starbasessd, 2020-12-14:
Might not help in your case, if you can't force camera back on via command,
as it may have come up, and gone back to sleep before you switch to it. If
it still records, though, that gives me a couple of options:
Replace the camera with a better behaved one.
restart motionEye every x seconds/minutes/hours
motionEyeOS has a couple of watchdogs, will try to see if it can watch my
rtsp feed later today.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-12-14:
No joy. It reboots the whole thing.
Will look at some other options.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-12-14:
Under Motion Notifications, turn on Run a Command
enter:

systemctl restart motioneye

Apply the changes
Reboot
Let me know if this helps.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Damian411, 2020-12-14:
Good idea. The live feed comes back with a short delay due to the restart time. 
But it restarts the process every 10-20s, I do not why, motion gap is configured to 30s.

starbasessd, 2020-12-14:
30 seconds between motion detection events. I have mine set to 2 seconds,
in case something else happens almost immediately... Change to what makes
sense to you. Recording should continue as before, though.
All 4 settings are changeable to what makes sense to you.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-12-18:
Following up, did this resolve the issue?

Damian411, 2020-12-18:
It works, but with a delay and motioneye will be rebooted serveral times, so the live feed disrubt several times for a few seconds.

starbasessd, 2020-12-18:
Because the motionEye loses connectivity with the camera when it goes to
sleep, restarting the motionEye is about the only solution I can see. If
you restarted motion or the entire pi, you'd lose whatever was being
recorded, too. Since restarting the camera isn't a possibility (for the 2
reasons of you can't find such a command, and you want the battery life
extended by having the camera go to sleep) it would be a feature request or
enhancement to ignore loss of connectivity to the camera stream for
motionEye.
I'll leave this here for @jasaw and @ccrisan.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Damian411, 2021-01-17:
I have tried the latest Release of Motion (not Motioneye). There the stream in the WebUI restarts automatically,  when the RTSP comes back. So it seems that there is a solution for that issue.

starbasessd, 2021-03-05:
If you are still having issues, please open a new issue.

closed 2021-03-05 by starbasessd