motioneye-project/motioneye #1971: Kuberenetes examples

vd-rd, 2020-12-18:
Hello,
I have been running Motioneye on Kuberenetes for the last 6 months with no issues. If you are interested I can contribute to the wiki with instructions on how to get going.
Regards,
Vadim.

starbasessd, 2020-12-18:
Feel free to send it my way. Be sure to include:
Base OS
Version and install requirements for Kubernetes.
Which management package used.
I will duplicate and confirm prior to posting.

vd-rd, 2020-12-18:
> Feel free to send it my way. Be sure to include:
> Base OS

My cluster is running atop CentOS 8 but is should work the same regardless of the distro used as there are no interactions with the host OS.

> Version and install requirements for Kubernetes.

Any recent enough version should qualify. It has been running and validated against 1.18 through 1.20.

> Which management package used.

I didn't create a Helm chart or anything like this for motioneye, though it would be helpful. I have just handcrafted some manifests which provide basic requirements to function:
- storage (PV for settings and media)
- ingress (to expose web ui to internet)
- service 
- deployment

I will brush them up a bit and link the repo with example and explanations.

starbasessd, 2020-12-18:
Building an Ubuntu 20.04 Desktop VM for use (4 processors, 4GB RAM, 50 GB HD) for testing, when you get something posted. You can contact me direct starbasessd at gmail if you don't to post anything publicly. I am US ET UTC-5.

starbasessd, 2020-12-18:
Are you running Docker under/alongside Kubernetes?
What level servers are you using? CPU, RAM, Storage?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-12-18:
Have you tried running the Docker version? What issues did you have with
it? (would run at the same level as the Kubernetes supervisor, thus not
having the extra layer involved).

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-12-18:
Currently up and ready, I think.
VBox VM: 4 processors, 4GB, 50GB.
Ubuntu 20.04, fully updated.
Docker 19.03.8 build afacb8b7f0
kubectl v1.20.1
minikube 1.16.0

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

vd-rd, 2020-12-19:
Let's treat them one by one:

> Are you running Docker under/alongside Kubernetes? What level servers are you using? CPU, RAM, Storage?

I am not using Docker as a CRI, i am relying on CRI-O under Kubernetes. (Docker became deprecated with K8s 1.20). I suggest you look for another CRI (containerd, cri-o, kata etc)
For the application i have one deployment/camera with an additional motioneye acting as aggregator.

> Have you tried running the Docker version? What issues did you have with it? (would run at the same level as the Kubernetes supervisor, thus not having the extra layer involved).

I did try the Docker version a while ago and i had no issues with it. However, since i migrated all my workloads to Kubernetes it made sense to include this as well so i can benefit from centralized management, observability and so on.

Now for the instructions.
First thing i recommend is setting up a namespace for the Motioneye instance/s:

```
kind: Namespace
apiVersion: v1
metadata:
  name: app-cctv 
```
Depending on what storage provider you have in the cluster ([CSI](https://kubernetes.io/blog/2019/01/15/container-storage-interface-ga/)) you might want to rely on PV provisioning or simple emptyDir mounts ( think of docker volume mounts) if not. For more examples based on individual setups check [this out](https://kubernetes.io/docs/concepts/storage/volumes/). If PV is available:
(skip this step if you don't have a CSI)
```
# one PV is created for media storage (movie/picture)
kind: PersistentVolumeClaim
apiVersion: v1
metadata:
  name: motioneye-cam-storage-pv
  namespace: app-cctv
  labels:
    app: motioneye
    environment: prod
    project: cctv
spec:
  accessModes:
    - ReadWriteOnce
  resources:
    requests:
      storage: 100Gi
  storageClassName: storage-slow-apps
  volumeMode: Filesystem
---
# one PV is created for settings (as an enhancement ConfigMaps can be used instead)
kind: PersistentVolumeClaim
apiVersion: v1
metadata:
  name: motioneye-cam-settings-pv
  namespace: app-cctv
  labels:
    app: motioneye
    environment: prod
    project: cctv
spec:
  accessModes:
    - ReadWriteOnce
  resources:
    requests:
      storage: 20Mi
  storageClassName: storage-slow-apps
  volumeMode: Filesystem
```
A service needs to be created so the pod can be reached via a constant dns name:

```
kind: Service
apiVersion: v1
metadata:
  name: motioneye-service
  namespace: app-cctv
  labels:
    app: motioneye
spec:
  ports:
    - name: webui
      protocol: TCP
      port: 8765
      targetPort: 8765
    - name: stream
      protocol: TCP
      port: 8081
      targetPort: 8081
  selector:
    app: motioneye
  type: ClusterIP
```

Right now we are ready for the actual deployment of an instance of motioneye.

```
kind: Deployment
apiVersion: apps/v1
metadata:
  name: motioneye
  namespace: app-cctv
spec:
  replicas: 1
  selector:
    matchLabels:
      app: motioneye
  template:
    metadata:
        app: motioneye
        environment: prod
        project: cctv
    spec:
      volumes:
        - name: localtime
          hostPath:
            path: /etc/localtime
            type: File
        - name: motioneye-settings-pv-storage
          # if PVC are used, the next two lines can be uncommented
          # persistentVolumeClaim:
          #   claimName: motioneye-cam-settings-pv
          # if emptyDir is used uncomment the next line
          # emptyDir: {}
        - name: motioneye-settings-pv-storage
          # if PVC are used, the next two lines can be uncommented
          # persistentVolumeClaim:
          #   claimName: motioneye-cam-storage-pv
          # if emptyDir is used uncomment the next line
          # emptyDir: {}
      containers:
        - name: motioneye-container
          image: 'ccrisan/motioneye:master-amd64'
          ports:
            - name: stream
              containerPort: 8081
              protocol: TCP
            - name: server
              containerPort: 8765
              protocol: TCP
          resources: 
            requests:
              cpu: 0.5
              memory: 512Mi
            limits:
              cpu: 1
              memory: 1Gi
          volumeMounts:
            - name: localtime
              mountPath: /etc/localtime
            - name: motioneye-settings-pv-storage
              mountPath: /etc/motioneye
  strategy:
    type: Recreate
```
Finally, if an [ingress controller](https://kubernetes.io/docs/concepts/services-networking/ingress-controllers/) is available and you have also a TLS provisioner in place (for instance [cert-manager](https://cert-manager.io/)) you can expose motioneye to the outside world.
```
kind: Ingress
apiVersion: networking.k8s.io/v1
metadata:
  name: motioneye-ingress
  namespace: app-cctv
  annotations:
    cert-manager.io/cluster-issuer: cloudflare-letsencrypt-issuer
    kubernetes.io/ingress.class: nginx
spec:
  tls:
    - hosts:
        - motioneye.example.com
      secretName: motioneye-prod-tls
  rules:
    - host: motioneye.example.com
      http:
        paths:
          - path: /
            pathType: ImplementationSpecific
            backend:
              service:
                name: motioneye-service
                port:
                  name: webui
```

starbasessd, 2020-12-19:
Can / would you, please create a basic step-by-step install text document
like
https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29
I then can test and modify to fit. This would need to include instructions
for installing Kubernetes, whichever CRI-O you've tested with.
The default kubernetes in Ubuntu yesterday (12.17.2020) installed Docker by
default on Ubuntu 20.04 LTS.
So this may be a distro issue.
How will your setup deal with EoE (End of Existence)  of python2.7 in 2
weeks?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

vd-rd, 2020-12-19:
> Can / would you, please create a basic step-by-step install text document like https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29

Yes, i will.

> This would need to include instructions for installing Kubernetes, whichever CRI-O you've tested with.

The installation of Kubernetes is out of scope IMO. Also, the underlying CRI used makes no difference. So in that regards, it doesn't matter if i used CRI-O and you used Dockershim. To conclude, there is no need to provide instructions for CRI or Kubernetes setup. The intended audience has already notion of cluster setup / or a cluster running, if it is searching for motioneye setup instructions for Kubernetes, right?

> How will your setup deal with EoE (End of Existence)  of python2.7 in 2 weeks?

I haven't taken this into account, but from my perspective the container image will continue to work. This is more of an issue for the project itself if it relies on Python 2.7.

starbasessd, 2021-03-05:
If you are still having issues, please open a new issue.

closed 2021-03-05 by starbasessd